evilsocket/pwnagotchi #816: [BUG] in /pwnagotchi/plugins/default/grid.py for numerical SSID in exclusion config

k4nfr3, 2020-01-27:
**Describe the bug**
Hi,
i had issues reporting with Pwngrid. I checked and the issue is in function is_excluded(self, what):
If in the config file, you added a numerical SSID, the code sees the variable as INT and not string, and therefor throws an error in line 60
skip = skip.lower()

**To Reproduce**
Steps to reproduce the behavior:
1. Add in config 
    plugins:
        grid:
            enabled: true
            report: true # don't report pwned networks by default!
            exclude: # do not report the following networks (accepts both ESSIDs and BSSIDs)
                - 12345

2. you will find in the logs : [2020-01-27 11:57:34,768] [ERROR] [grid] error while checking pcaps: 'int' object has no attribute 'lower'

I'm no dev guy and have not much python experience.
Maybe we could add a simple type test
something along those lines 
def is_excluded(self, what):
        for skip in self.options['exclude']:
            **if (type(skip)==str) and (type(what)==str):**
                skip = skip.lower()
                what = what.lower()
            if skip in what or skip.replace(':', '') in what:
                return True
        return False

xenDE, 2020-01-27:
better use " in config for all/complex SSIDs you want to ha dle as string.

```
exclude: # do not report the following networks (accepts both ESSIDs and BSSIDs)
  - "12345"
```

or you can run into trouble with SSIDs like "00000"=(0) "0xff"=(255) "iLove#tags" (iLove) and so on

dadav, 2020-04-11:
use quotes

dadav, 2020-04-11:
Derek close

closed 2020-04-11 by derek[bot]