evilsocket/pwnagotchi #827: [Question] counter reset

frankenstein91, 2020-02-24:
I'm not sure if this is what the system intended or if it's a bug.
If I load the handshakes from the system and delete them, the counter on the display is reset.
So I would have to keep all files on the system to be able to keep eternal statistics. Especially since my system was not yet online.
Could someone confirm for me that I am not the only one and if it is a bug?

Thank you

FractumSeraph, 2020-03-10:
It looks like it just counts the number of .pcap files in that directory to get that number.

https://github.com/evilsocket/pwnagotchi/blob/ccea7cbeefe3c65a2af8fd706a5a2d08b93ea15e/pwnagotchi/utils.py#L171

frankenstein91, 2020-03-14:
would be nicer per sqlite