evilsocket/pwnagotchi #829: [BUG] Web UI changes not saved - Rp 3B+

mgagnequebec, 2020-03-02:
**Describe the bug**
On the rasperry pi 3b+, after a fresh install + update, the web ui won't save changes I make. 

( my fresh install booted without lan cable pluged in the first time, not sure if it matters)
**To Reproduce**
now In MANU mode:

1. I accessed the web ui through 192.168.0.102:8080 
2. Went to plugin tab. 
3. Enabled the webcfg (couldnt click it until I switched to another tab and came back) 
4. Made changes (added bt mac and set option for this to true) 
5. Saved my changes
6. Putty ssh -> 192.168.0.102:22
7. login
8. sudo su
9. nano /etc/pwnagotchi/config.yml

**Expected behavior**
Changes saved to config.yml

**Environment (please complete the following information):**
 - Pwnagotchi version (pwnver is broken os not sure, but I just did a git pull + pip3 install) 
 - Raspberry pi 3b +

xBelladonna, 2020-03-02:
If you pulled latest source from the repo you will be using toml config not yaml. Edit `/etc/pwnagotchi/config.toml` instead.

mgagnequebec, 2020-03-03:
Ah gotcha, I see the thread about it now. Thanks for the info, I'll see if I can get bt-tether enabled in the toml then

LowSkillDeveloper, 2020-03-03:
This means that the webcfg plugin is not rewritten for the new format, i think.
Switching to toml will create many problems and misunderstandings at the beginning. Need to change not only the instructions for users, but also rewrite the plugins...

mgagnequebec, 2020-03-03:
It's actually much easier with toml, even without the web ui working. So I'd say this change is a plus.

This project is so amazing, I wish I could contribute more but I am still very new to the project. I will test as much as I can though.

dadav, 2020-04-04:
Derek close

closed 2020-04-04 by derek[bot]