evilsocket/pwnagotchi #830: Support for Color e-ink displays/Theme Support

harleyknd1, 2020-03-05:
**Is your feature request related to a problem? Please describe.**
No, although I would personally find a lack of colors a problem. It's not a functional problem.

**Describe the solution you'd like**
It would be nice/fun if the pwnagotchi allowed you to set custom colour/theme profiles in the configuration. Enabling users to use their colored e-ink displays to their full potential.

I'm aware that you can flip the BLACK and WHITE variable (0x00 and 0xff) around to get an inverted appearance. But I think it would be rather nice to be able to be able to tweak this as well as colour without having to adjust the python files themselves.

e.g:
`Theme:
             Face: RED
             Text: WHITE
             Background: Black
             Plugins:
                          UPS-Lite: Red`

**Describe alternatives you've considered**
An alternative would require the adjustment of the python files themselves, which might break with each update of the pwnagotchi

**Additional context**
This probably isn't super high on the to-do list, since it's a novel thing if anything. But I still like colour in my life dammit!
[image]

syedrali, 2020-03-12:
Related to this, I have a red, white, black inky phat as shown above. In /usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/view.py where you can invert the colours by swapping round the values for BLACK and WHITE, could I change the value for say BLACK to actually be red? If so what hex code should be used?

kruffin, 2020-03-31:
Took a quick look and it will probably require another drawing surface for the additional color:

https://github.com/evilsocket/pwnagotchi/blob/8260b41bab84c2583f2f82f50f1d97f2ebc9ac46/pwnagotchi/ui/hw/waveshare213d.py#L43
https://github.com/evilsocket/pwnagotchi/blob/8260b41bab84c2583f2f82f50f1d97f2ebc9ac46/pwnagotchi/ui/hw/libs/waveshare/v213bc/epd2in13bc.py#L311

The display() method of these tri-color displays can take an additional image for the other color. This however, means you wouldn't be able to just send a different hex value since those are also used for grayscale (white [0xff] to black [0x00] gradient).