evilsocket/pwnagotchi #834: Unsupported display waveshare_32

kenn850, 2020-03-12:
I booted up pwnagotchi in raspberry Pi 3 Model B+ with a Waveshare 3.2Inch LCD RPi.
It is eating handshakes and I am able to retrieve them, no issue with that. The only issue is that the display is unsupported. I tried changing display type to 'inky' and 'inkyphat', both didn't work. By doing so the only thing that happened is that the display turned whole white. Some flashes come and go but nothing clear.

oglok, 2020-03-22:
@kenn850 could you please share the procedure you follow to configure the screen and pwnagotchi?

Thanks

lildaze, 2021-04-04:
> @kenn850 could you please share the procedure you follow to configure the screen and pwnagotchi?
> 
> Thanks

hey I'm having some issues with my setup to on a zero w with a inkphat v2 can't seem to get to display anything here's my config file if u can help that'll be coo 

main.name = "pwnagomoni"
main.lang = "en"
main.whitelist = [
  "Pretty Fly for a Wi-Fi"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "Pretty Fly for a Wi-Fi"
]

ui.display.enabled = true
ui.display.type = "inky"
ui.display.color = "black"