evilsocket/pwnagotchi #837: Frequent AI Error While Training Reported in Logs

syedrali, 2020-03-15:
In the logs, I am seeing AI error while training frequently:

```
[2020-03-14 23:44:07,821] [ERROR] [ai] error while training
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tensorflow_core/python/client/session.py", line 1348, in _do_call
    return fn(*args)
  File "/usr/lib/python3/dist-packages/tensorflow_core/python/client/session.py", line 1333, in _run_fn
    target_list, run_metadata)
  File "/usr/lib/python3/dist-packages/tensorflow_core/python/client/session.py", line 1421, in _call_tf_sessionrun
    run_metadata)
tensorflow.python.framework.errors_impl.UnimplementedError: Cast string to float is not supported
         [[{{node RMSProp/update_model/vf/w/Cast}}]]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 242, in learn
    self.num_timesteps // self.n_batch, writer)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 210, in _train_step
    [self.summary, self.pg_loss, self.vf_loss, self.entropy, self.apply_backprop], td_map)
  File "/usr/lib/python3/dist-packages/tensorflow_core/python/client/session.py", line 941, in run
    run_metadata_ptr)
  File "/usr/lib/python3/dist-packages/tensorflow_core/python/client/session.py", line 1164, in _run
    feed_dict_tensor, options, run_metadata)
  File "/usr/lib/python3/dist-packages/tensorflow_core/python/client/session.py", line 1342, in _do_run
    run_metadata)
  File "/usr/lib/python3/dist-packages/tensorflow_core/python/client/session.py", line 1362, in _do_call
    raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.UnimplementedError: Cast string to float is not supported
         [[node RMSProp/update_model/vf/w/Cast (defined at /lib/python3/dist-packages/tensorflow_core/python/framework/ops.py:1692) ]]
```

Is something wrong with my set-up or configuration? Is it anything to be concerned about?

dadav, 2020-04-04:
What is the config value of `ai.epochs_per_episode`?

dadav, 2020-04-12:
Derek close

closed 2020-04-12 by derek[bot]