evilsocket/pwnagotchi #844: Pimoroni led shim notifications

bking46, 2020-04-03:
**Is your feature request related to a problem? 
Please describe.**
Yes, unable to install/use pimoroni led shim.

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I would like to make use of a pimoroni led shim I
have laying around to display led notifications. For example when a handshake is captures, or when pwnagotchi is scanning.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
---
To enable install and use of pimoroni led shim for more customisation of the pwnagotchi. Below is the git link to the library and everything that is needed, including examples:
 https://github.com/pimoroni/led-shim
Also the hardware is v easy to use, it plugs directly on the GPIOs and doesn't interfere with the eink paper display as far as I am aware

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
There is already a pin for gpio buttons so I think this should be to time consuming to implement.

**Additional context**
Add any other context or screenshots about the feature request here.

bking46, 2020-04-03:
Incorrect folder permissions didn't allow me to run the python code. Would be nice if we can have this integrated to show all sorts of notifications 😊. Attached image for reference.
[image]

kbwrx, 2020-09-16:
Bump... This would be awesome.

kbwrx, 2020-09-16:
I happen to have one laying around if anyone needs me to test. Thx

innuendoz, 2023-04-11:
i was hoping this got more attention