evilsocket/pwnagotchi #856: DFRobot V2 support

misterf13, 2020-04-17:
**Is your feature request related to a problem? Please describe.**
It seems that only dfrobot e-ink v1 is supported. The v2 screens use a new library it seems. https://github.com/DFRobot/DFRobot_RPi_Display_V2 instead of https://github.com/DFRobot/DFRobot_RPi_Display

**Describe the solution you'd like**
Maybe have dfrobot support like the waveshares. dfrobot_1 and dfrobot_2

**Describe alternatives you've considered**

**Additional context**
I got it to work, but not sure how clean this is, etc.
https://github.com/misterf13/pwnagotchi/commit/faddca544057fd6b1448b32fe489d8e8a1f2cbf9

dadav, 2020-04-18:
looks good to me, cherry-picked it

closed 2020-04-20 by evilsocket

nimbo78, 2020-04-23:
@dadav 
config.toml:
```
ui.display.enabled = true
ui.display.type = "dfrobot"
ui.display.color = "black"
```

```
root@pwnagotchi:/home/pi# pwnagotchi
unsupported display type dfrobot
```
am i doing something wrong?

misterf13, 2020-04-23:
> @dadav
> config.toml:
> 
> ```
> ui.display.enabled = true
> ui.display.type = "dfrobot"
> ui.display.color = "black"
> ```
> 
> ```
> root@pwnagotchi:/home/pi# pwnagotchi
> unsupported display type dfrobot
> ```
> 
> am i doing something wrong?

Try "dfrobot_1" or "dfrobot_2" depending on what version you have.

nimbo78, 2020-04-23:
> 
> 
> > @dadav
> > config.toml:
> > ```
> > ui.display.enabled = true
> > ui.display.type = "dfrobot"
> > ui.display.color = "black"
> > ```
> > 
> > 
> > ```
> > root@pwnagotchi:/home/pi# pwnagotchi
> > unsupported display type dfrobot
> > ```
> > 
> > 
> > am i doing something wrong?
> 
> Try "dfrobot_1" or "dfrobot_2" depending on what version you have.

yeap, now it works :) thnx a lot
may be someone could edit this instructions - https://pwnagotchi.ai/configuration/#initial-configuration ?

nimbo78, 2020-04-23:
> 
> 
> > @dadav
> > config.toml:
> > ```
> > ui.display.enabled = true
> > ui.display.type = "dfrobot"
> > ui.display.color = "black"
> > ```
> > 
> > 
> > ```
> > root@pwnagotchi:/home/pi# pwnagotchi
> > unsupported display type dfrobot
> > ```
> > 
> > 
> > am i doing something wrong?
> 
> Try "dfrobot_1" or "dfrobot_2" depending on what version you have.

also, is there any chance to use 2 buttons? with gpio_buttons plugin or something alse could help me?

misterf13, 2020-04-23:
In your main config you probably need the following.

```
gpio_buttons:
    enabled: true
      gpios:
        21: 'some command'
        22: 'some other command'
```

Check out the code here for more info. :)

https://github.com/evilsocket/pwnagotchi/blob/7de512103345454ff6ff168cf04a7b476b4bb090/pwnagotchi/plugins/default/gpio_buttons.py#L29

PR #855 (2020-04-20): Bug fixes
- fixed `asyncio bug`
- fixed bt-tether so that its `useable in manu-mode`
- added `tooltips` to plugin page (problem: description can only be show if plugin is loaded/enabled)
- added `Watchdog-Plugin` (reloads wifi driver on blind bug)
- added `bash_completion` script
- added `DFRobot V2 support` (closes #856)
- added wpa-sec/ohc dashboard redirection in /plugins
- added `main.custom_plugin_repos` option