evilsocket/pwnagotchi #859: DNS not working

careyer, 2020-04-20:
**Describe the bug**
When using internet connection sharing (Windows 10) I cannot get DNS working. Pwnagotchi always tells me "ping: google.com: Temporary failure in name resolution." when I try to ping google for example. I can ping the direct IP though, so internet connection sharing is in principle working.

I can temporarily fix the problem by editing "/etc/resolv.conf" and replace 127.0.0.1 with 8.8.8.8 in there. It will instantly start working after that. However the resolv.conf will be overwritten on each boot with 127.0.0.1 again.

Also tried setting a DNS server in the RNDIS Ethernet Interface in windows (8.8.8.8) but that has no effect. Tips are welcome!

Thanks!

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.3
 - Windows 10
 - PC

agent932, 2020-04-21:
I have had this same issue for the past few releases

careyer, 2020-04-21:
Former release 1.4.3 was working well for me. Re-Imaged 1.5.3 several times now but the problem persists. Also tryied  a fresh 1.5.3 install and 1.5.3 with restore from backup. Both variants have the same problem. Reverted back to 1.4.3 and there it works without problems.

dadav, 2020-04-22:
Derek assign: dadav

dadav, 2020-04-22:
As far as I know, we didnt rly change anything related to dns. But we used a new raspbian, so maybe something changed there

dadav, 2020-04-22:
Omg, dnsmasq!!?

nimbo78, 2020-04-23:
@dadav 
when trying to connect though inet sharing with win10:
```
cat /etc/resolv.conf
# Generated by resolvconf
nameserver 127.0.0.1
```
for some reason now rpi can't get dns from win10..
may be we need use some static nameservers?

dadav, 2020-04-23:
try: systemctl disable dnsmasq

then reboot

LowSkillDeveloper, 2020-05-19:
v1.5.3 still having problems with DNS.
"systemctl disable dnsmasq" didn’t help :(

rileyinman, 2020-06-24:
@LowSkillDeveloper I was looking at the bt-tether code for other reasons and remembered your message, because it looks like it was added in #279. I'm not exactly sure why this is the case... maybe @dadav could clarify.

LowSkillDeveloper, 2020-06-24:
Yes! This is it. When i setting up, I turned ON "share_internet" to unload handshakes.
And it seems DNS overlapped.
I'm not sure, but I think they should hit on different lines. There was no gap between them at all, in the screenshot above it can be seen.

rileyinman, 2020-06-24:
It does seem odd that it ended up after your current DNS settings, considering it's supposed to `seek(0)` and write with a newline (meaning it would end up on a line above all other configuration). However, I'm wondering why it's even necessary to override the user's DNS configuration.

ByerRA, 2020-09-09:
Has any progress been made on this issue as I just had to re-flash my pwnagotchi due to a bad SD card and are running into the same issue.  If I manually edit the /etc/resolv.conf file it doesn't solve the problem and disabling dnsmasq doesn't work either.

Gizmo1k, 2020-09-30:
i am also having this same issue.

normsteel, 2020-10-06:
Adding "server=8.8.8.8@usb0" to /etc/dnsmasq.conf seems to help.

meldraknailo, 2020-10-07:
I've had this issue since setting up my panagotchi about a month ago. Several reflashes later and I still have to edit resolv.conf every time I connect the unit to my pc or laptop (both windows 10).

meldraknailo, 2020-10-07:
I found a threat that allowed  permanent fix:  
add: "server=8.8.8.8@usb0" to /etc/dnsmasq.conf
That fixed my issue and I automatically share connection on windows now.

hschult, 2020-10-07:
Neither disabling dnsmasq nor adding "server=8.8.8.8@usb0" did work for me.

ByerRA, 2020-10-07:
I an also confirm that adding "server=8.8.8.8@usb0" didn't work for me.

davmol, 2020-10-17:
Any new ideas about a solution? I'm really struggling to get it to work...

syzspectroom, 2020-11-08:
> 
> 
> Adding "server=8.8.8.8@usb0" to /etc/dnsmasq.conf seems to help.

fixed the issue for me.

TheCrickler, 2020-11-27:
Try adding `dns-nameservers 8.8.8.8` under the gateway line in `/etc/network/interfaces.d/usb0-cfg` 

The _resolveconf_ **program** is responsible for auto-generating the _resolve.conf_ **file** during boot based on the files in `/etc/network/interfaces.d/`

Since raspbian is based on debian, see [Debian Wiki: Defining The Nameservers](https://wiki.debian.org/NetworkConfiguration#Defining_the_.28DNS.29_Nameservers)

It has been working stable for me, I hope this helps.

arn033, 2020-12-30:
> Try adding `dns-nameservers 8.8.8.8` under the gateway line in `/etc/network/interfaces.d/usb0-cfg`
>
> The _resolveconf_ **program** is responsible for auto-generating the _resolve.conf_ **file** [...]

Hello, fixed the issue for me.
Great job.
Thanxs

Spessmen14, 2021-02-26:
None of the solutions working for me , but I have Win7

evilsocket, 2021-04-24:
hey @dadav nice to see you active again! will you send a PR for this or should i fix it?

evilsocket, 2021-04-24:
@dadav very nice of you forking the repo, using the issues here to fix your own fork and not sending a PR back 👍🏻

closed 2021-04-24 by evilsocket

meldraknailo, 2021-09-13:
So I have been having issues with this forever, but finally seem to have it worked out such that any of my units will auto connect to the web when I plug them in without going through a bunch of crud. I use windows 10:

Connect the unit via usb.

Go to network adapter settings, ipv4, and set up the pwnagotchi as 10.0.0.1, 255.255.255.0, 8.8.8.8

Go to settings of the network you use to connect to the web. Unshare then share that connection with the pwnagotchi. Then go back to check the ipv4 for the pwnagotchi and make sure they are correct.

Should be able to ssh into the unit after that and use WinSCP or putty to find dnsmasq.conf and resolv.conf in /etc/.

I changed resolv.conf to contain only: nameserver 8.8.8.8nameserver 8.8.4.4nameserver 9.9.9.9

I then disabled dnsmasq using: systemctl disable dnsmasq

My dnsmasq.conf now says only:

interface=br0

expand-hosts

domain=example.com

dhcp-range=192.168.20.100,192.168.20.150,12h

databonanza, 2022-06-16:
> @dadav very nice of you forking the repo, using the issues here to fix your own fork and not sending a PR back 👍🏻

It seems that issues seem to go unfixed for long periods of time.  To me, this indicates that the lead developer is too busy to maintain the project anymore.

Why would he send a PR back if it doesn't seem like you're trying very hard to "service" the community that is trying to use a product you made?  After dealing with the issue of not being able to go into AI mode for days now, I've realized that there isn't much interest in resolving this issue.  The newest image of pwnagotchi is broken and wont' go into AI mode.  People have opened bugs about it.  Looks like some are over a year old now without a viable fix.

evilsocket, 2022-06-16:
@databonanza  

> To me, this indicates that the lead developer is too busy to maintain the project anymore.

or maybe he trusts the community, that already did an amazing job in contributing, to help with the fixes given he has literally dozens of other projects he's working on ... who knows :D

>  ... if it doesn't seem like you're trying very hard to "service" the community ...

oh, i didn't know that who's releasing unpaid software used by hundreds of people for free has an obligation to "service" the community, especially considering that such community is filled with other developers that can contribute with their code, fixes and PRs that get periodically merged ... noted!

>  trying to use a product you made?

this is not a product, it's free software.

> After dealing with the issue of not being able to go into AI mode for days now, I've realized that there isn't much interest in resolving this issue.

wow you're full of realizations and assumptions! i wish i as informed as you seem to be about this whole thing :D

evilsocket, 2022-06-16:
one might also argue that the more people with your attitude are around (and unfortunately there're plenty), the less one is willing to actually work on anything they're asking for

databonanza, 2022-06-16:
^^ are you suggesting that you haven't fixed this because you have a community of people "with the wrong attitude"?  Or is it something more normal like, you're too busy with other things?  The former would be an incredibly poor choice to have people believe.

evilsocket, 2022-06-16:
let's say 90/10, but you're free to believe whatever you want :D

adios

databonanza, 2022-06-16:
I bet with all the time spent thinking out the replies here you could have made a patch to fix the not going into AI issue today.

commit 298ed240 (2021-04-24): fix: fixes #859 (thanks @dadav for not bothering to contribute to the main repo)