evilsocket/pwnagotchi #861: dfrobot v2 ui gps is not show in the right location of the screen

misterf13, 2020-04-21:
**Is your feature request related to a problem? Please describe.**
Not a big problem, just want to print out gps data on a better location of the screen.

**Describe the solution you'd like**
I made changes and tested them on my pwnagotchi. Submitted in https://github.com/evilsocket/pwnagotchi/pull/860

Thank you.

elipsius, 2020-05-26:
I noticed mine looks awful too when GPS coordinates are being displayed (attaching image)

[image]

misterf13, 2020-06-03:
The PR got merged so I am sure it will be fixed next release. You can also clone master right now, install it, and see if it is working out for you. 👍