evilsocket/pwnagotchi #862: [BUG] issue of config file since upgrade to 1.5.3

k4nfr3, 2020-04-21:
**Describe the bug**
Error message list of index out of range in debug mode. I even copied the default.yml to config.yml but it doesn't solve the issue.
Any suggestions ?

pwnagotchi --version
1.5.3
pwnagotchi --print-config
ERROR:root:There was an error processing the configuration file:
list index out of range
which pwnagotchi
/usr/local/bin/pwnagotchi
ls -la /usr/local/bin/pwnagotchi
-rwxr-xr-x 1 root root 5596 Apr 21 20:11 /usr/local/bin/pwnagotchi
cp default.yml config.yml
pwnagotchi --print-config
ERROR:root:There was an error processing the configuration file:
list index out of range

**To Reproduce**
not sure if it happens to other people

**Environment (please complete the following information):**
 - Pwnagotchi version
 - OS version
 - Type of hardware : standard Pi Zero

dadav, 2020-04-22:
Hi, thanks for your report.

We changed the configuration language to `toml`. The `defaults.yml` doesn't works anymore and will be ignored.

Therefore, if your configuration directory contains a `config.toml`, your `config.yml` will be ignored too.
Please check the `config.toml` for errors^^

k4nfr3, 2020-04-22:
hello, 
thanks for that. Indeed it's there the problem.
Question, the upgrade has converted my config.yml to config.toml but I see a big difference of structure between my config.toml and the default.toml
my converted config.toml contains 

[main]
name = "k4nfr3g0tchi"
lang = "en"
custom_plugins = "/root/pwnagotchi-plugins-contrib"

and the default.toml has
main.name = ""
main.lang = "en"
main.confd = "/etc/pwnagotchi/conf.d/"

1) What is the right format or does it matter ?
2) is there an issue in the upgrade tool if the result has a wrong format output ?

Regards
K4nfr3

k4nfr3, 2020-04-22:
I've changed to the format provided in default.toml and all works fine now.
Case close. Not sure if anybody want to look into the upgrade process script to adapt to the new format. (not sure)

closed 2020-04-22 by k4nfr3

dadav, 2020-04-22:
Maybe you want to delete your current config.toml and convert it again. There was a "style" change. But both works