evilsocket/pwnagotchi #863: [BUG]

ReconPirate, 2020-04-23:
**Describe the bug**
The bug in question is my pwnagotchi skipped from 1.5.1 to 1.5.3 via auto update, and now seems to pwn no networks, it still deauths and caused handshakes to pass in it's range by force via WiFi pineapple and still nothing. 
**To Reproduce**
Steps to reproduce the behavior:
1 be on 1.5.1 and auto update 
**Expected behavior**
Still associates still deauths, shows so in dmesg, but nothing.  Any clues before I do a clean wipe?

colossus700, 2020-04-23:
I had the same issue and doing a full reinstall and reloading the handshakes/brain from the backup fixed it.

miotislucifugis, 2020-04-23:
Im in the same boat... was hoping to avoid a full reinstall. :(

dadav, 2020-04-24:
Please try: `bettercap -eval "caplets.update;q"` and reboot

miotislucifugis, 2020-04-24:
That seems to have worked for me.  Phew!
Thanks!

ReconPirate, 2020-04-24:
Seems to have fixed my issues as well.  Thanks for the help :3

closed 2020-04-24 by ReconPirate