evilsocket/pwnagotchi #864: bettercap GUI - 401 Unauthorized: wrong credentials

OfficialMuffin, 2020-04-24:
## Expected Behaviour
Login using 'changeme' as the username and 'changeme' for the password

## Current Behaviour
Displays error 401 and unable to use the webui

## Your Environment

- [X] You're using the official images

- [X] You're using a raspberry pi 0 w

- [ ] You're using a supported LCD

I have followed the instructions up to this [point](https://pwnagotchi.ai/usage/#user-interface) but I am unable to use the GUI.
This is a freshly installed [v1.5.3](https://github.com/evilsocket/pwnagotchi/releases/tag/v1.5.3)
I also have .ssh/authorized_keys file with pubkey authentication enabled on the Pi

## config.toml Contents 

```
main.name = "Pwnagotchi"
main.lang = "en"
main.whitelist = [
  "WiFi1",
  "WiFi2",
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "WiFi1"
]

ui.display.enabled = true
ui.display.type = "waveshare_2"
ui.display.color = "black"

```

natur3kid, 2020-04-25:
The default credentials are pwnagotchi: pwnagotchi

epantke, 2020-04-26:
After updating bettercaps caplets you need to change your password and the username in the pwnagotchi caplets to whatever is in your "/etc/pwnagotchi/config.toml" file. 

"Only run caplets.update the first time as every time the entire system caplets folder is replaced with the downloaded contents from github, overwriting your changes, such as the credentials, with default values."

"The default authentication credentials are pwnagotchi:pwnagotchi, if you decide to change them in /usr/local/share/bettercap/caplets/pwnagotchi-*.cap, you’ll also need to update the configuration in /etc/pwnagotchi/config.toml to use the new credentials."

OfficialMuffin, 2020-04-27:
So what's the "changeme" link to? Im confused. Sorry

closed 2020-04-27 by OfficialMuffin

danielnegromar8732, 2026-10-21:
> 1. The default credentials are pwnagotchi: pwnagotchi

Arrived here looking for the credentials. In current version the default Bettercap username is user, and the default password is pass.
Source: https://www.stationx.net/bettercap-tutorial/