evilsocket/pwnagotchi #865: [BUG] Saving .pcaps with no handshakes.

FractumSeraph, 2020-04-26:
**Describe the bug**
My pwnagotchi is saving .pcap files and marking networks as pwned even if no valid handshake was captured. 

**To Reproduce**
Steps to reproduce the behavior:
1. Collect a few captures.
2. Open them in Wireshark.
3. Check if they are valid.
4. Notice that most of them are not.

**Expected behavior**
Pwnagotchi should use tools like aircrack/tshark to ensure that a cap contains a valid handshake.

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.5.3
 - Type of hardware: RPi0W
 - Any additional hardware used: No

**Additional context**
This did not happen in previous versions.

xenDE, 2020-04-26:
hi,

if your workflow needs full handshakes you can do this with an plugin.

or use/fork this plugin and change to your needs, if you only wan't aircrack confirmed handshakes:

https://github.com/evilsocket/pwnagotchi-plugins-contrib/blob/master/aircrackonly.py

That's not a bug imho, please close this issue if your question is hereby answered

best regards

FractumSeraph, 2020-04-27:
I guess I should also point out that these are not pmkids either.
When I get home I'll set up a test network and post my results including the .cap.

closed 2021-04-14 by FractumSeraph