evilsocket/pwnagotchi #866: Whitelist : Pwnagotchi deauth my device

Baallrog, 2020-04-27:
**Describe the bug**
In the toml file I have define my network BSSID ans SSID and still Pwnagotchi deauth my device on my network.

**To Reproduce**
No precise step, I fill the whitelist with my BSSID and SSID and let the Pwnagotchi run for a couple of minutes.

**Expected behavior**
The whitelist should be taken in account.

**Screenshots**
Nope

**Environment (please complete the following information):**
 - Pwnagotchi 1.5.3
 - OS version
 - Pi Zero WH

**Additional context**
In the log the Sonos device are mine.

[pwnagotchi.log](https://github.com/evilsocket/pwnagotchi/files/4541225/pwnagotchi.log)

Baallrog, 2020-04-27:
[config.txt](https://github.com/evilsocket/pwnagotchi/files/4541240/config.txt)

grokbeer, 2020-05-08:
Your config shows that you've whitelisted two SSID names but only one BSSID MAC (`00:xx:xx:xx:xx:17`).

Only some 802.11 frames contain the SSID so it's not unusual that pwnagotchi won't always know the SSID of an AP.  Your log shows that pwnagotchi didn't initially see your SSID when it first tried to associate and then deauth your Sonos:
<pre>
[2019-07-10 01:29:28,593] [INFO] sending association frame to  (00:xx:xx:xx:xx:16 Sagemcom Broadband SAS) on channel 1 [3 clients], -60 dBm...
[2019-07-10 01:29:37,669] [INFO] deauthing 78:xx:xx:xx:xx:22 (Sonos, Inc.) from  (00:xx:xx:xx:xx:16 Sagemcom Broadband SAS) on channel 1, -60 dBm ...
</pre>
(Note the two spaces after `to` and `from` which is where the SSID would show if it was known.)

You can see that the BSSID that pwnagotchi saw for your network was `00:xx:xx:xx:xx:16` and not `00:xx:xx:xx:xx:17` as you had whitelisted. Perhaps `:17` is the BSSID for your `_5` SSID?

It's important to whitelist all BSSIDs for your network since your SSID won't always be known to pwnagotchi. One easy way to do this is to leave off the last octet of the BSSID, e.g. `00:xx:xx:xx:xx`, which would mean that both `00:xx:xx:xx:xx:16` and `00:xx:xx:xx:xx:17` would be covered by a single whitelist entry.

It's common for the 2.4 GHz and 5 GHz radios in a single AP to have the same BSSID MAC prefix, but if you have multiple APs on your WLAN you should make sure to whitelist their BSSIDs too.

Baallrog, 2020-05-09:
Hi,

It was that !!
I totally forgot the thing that a dual router (5ghz/2.4ghz) is like two router ...

Thank you.

Baallrog, 2020-05-25:
Hi,

I think there is a problem. Maybe it's me.

I reboot the Pwnagotchi by plugging out/in because of a freeze.

And now my own wifi network is target.

There the whitelist : 
main.whitelist = [
  "00:37:b7:02:9b:17",
  "00:37:b7:02:9b:16",
  "SFR-9b11934_5",
  "SFR-9b11934"
]

Baallrog, 2020-05-25:
[image]
[image]

rezzorix, 2020-10-31:
Yes, I think the whitelist is not functioning correctly at the moment.

My Pwnagotchi version is: 1.5.3
I have whitelisted networks SSID & BSSID (Mac Address) - still get deauth.

It is understood, that handshakes will still be captured, but the constant deauths shouldnt happen...

qkum, 2021-03-18:
Imo the easiest fix is to turn deautch off. 

Easy to do 2021. 

in the web UI, use the in-build plugin that let you change the config file live :) (**webcfg** plugin)