evilsocket/pwnagotchi #868: [BUG] on_handshake never called

tonjo, 2020-04-30:
**Describe the bug**

I tried some plugins: at first `paw-gps`, enabled it, and noticed whenever a handshake is captured,
nothing happens, I would expect at least the logging (line 32 in `paw-gps.py` version 1.0.0).
I had previously installed and configured PawServer and did all the steps described in 
[https://gist.github.com/Arttumiro/d359769bc492559fe3098ac890a3c807](https://gist.github.com/Arttumiro/d359769bc492559fe3098ac890a3c807)

Also tried other contrib plugins, such as quickdic (installing `aircrack-ng` first, of course).
I would expect at least the `logging.info` lines to be executed, but nothing.
I created a plugin myself, with a simple logging, but nothing.
Other parts of such plugins work, though, for example inside `on_association`, `on_wifi_update`, etc.

**To Reproduce**
1. Use any plugin that implement `on_handshake`
2. Ensure deauth is enabled
3. `tail -f /var/log/pwnagotchi.log`

**Expected behavior**
At every handshake captured, I would expect the execution of callback `on_handshake`

**Environment:**
 - Pwnagotchi version: **v1.5.3**
 - RPI Zero W
 - Android Smartphone via bt-tether (no e-ink display)
- Ubuntu 18.04.04 via USB

**Additional context**
**Interesting: number of current session's pwned networks remains 0**, but it clearly cannot be true
because the total count increases. Also the **name of the network of most recent acquired handshake is empty**. It was once visibile.

_NOTE_:
I had version 1.0.1, then stopped for a while, then upgraded to version 1.4.3, by backupping, reflashing and restoring files.
Since than, current network pwned was 0.
After that I enabled auto_update, and now I'm on 1.5.3 as stated before.
I wonder if I've missed some steps.

Bobtron, 2022-07-13:
Same issue occurs here:
https://www.reddit.com/r/pwnagotchi/comments/vxu426/pawgps_issue_skipping_event_handshake_because_the/
Running 1.5.5 version

Bobtron, 2022-07-14:
Never mind, looks like the issue above was caused by a user error in paw-gps configuration