evilsocket/pwnagotchi #870: macOS connection share isn't working

samyhaff, 2020-05-06:
**Describe the bug**
The script doesn't share the wifi nor ethernet connection,
instead I get the following errors:

`
sharing connecting from upstream interface en0 to usb interface en5 ...
net.inet.ip.forwarding: 1 -> 1
No ALTQ support in kernel
ALTQ related functions disabled
pfctl: pf already enabled
pfctl: Use of -f option, could result in flushing of rules
present in the main ruleset added by the system at startup.
See /etc/pf.conf for further details.

No ALTQ support in kernel
ALTQ related functions disabled
`

**To Reproduce**
Steps to reproduce the behavior: run the script after being able to ssh 

**Environment:**
 - latest pwnagotchi version
 - macOS 10.15.4
 - MacBook pro 2017

miotislucifugis, 2020-05-06:
Thats normal.   I always get that and connection share works, but not always.   I think it is more of a mac thing not pwnagotchi.     Ive found that it will only work if my pwnagotchi in in manual mode and that I need to re-run the script sometimes after rebooting  my macbook.

samyhaff, 2020-05-06:
well that's strange, what's happening for me is that as soon as I launch the script I am even no longer able to ping my pwnagotchi.

miotislucifugis, 2020-05-06:
hmm. not sure what that could be .  Is it still being "seen" in network connections?   Im still running High Sierra, so it could be a new mac permissions thing.     Make sure that you've set up the Pwng to have a static RDNIS, like shown in the "community hacks"  section.   I dont know why that is enabled by default, but if it is changing every time the unit reboots, you will have connection sharing issues.

samyhaff, 2020-05-06:
Nevermind, I eventually got it to work thanks to help on slack.

closed 2020-05-06 by samyhaff