evilsocket/pwnagotchi #875: [BUG] Watchdog plugin constantly restarts application

grokbeer, 2020-05-08:
**Describe the bug**
The watchdog plugin does a good job of detecting the blindbug issue but since it only restarts the pwnagotchi service the kernel log doesn't change and so it constantly detects the same bug and restarts the applications over and over.
Does restarting the application actually fix this bug or is a reboot required?

**To Reproduce**
1. Enable the watchdog plugin.
2. Wait for 5 "brcmf_cfg80211_nexmon_set_channel.*?Set Channel failed" messages to appear in the kernel log in a 5-minute period.
3. Watch the pwnagotchi service restart continuously.

**Expected behavior**
The plugin should only restart the application once to fix the bug.

**Environment (please complete the following information):**
 - Pwnagotchi version: v1.5.3
 - OS version: Debian 10.3
 - Raspberry Pi Zero W

LowSkillDeveloper, 2020-05-18:
The same problem. But also the problem is that sometimes he sees blindness where it does not exist. Often the AI Mode does not have time to load up but already restart is triggered.

iiwwnnaa, 2021-03-27:
Same problem

closed 2021-04-18 by evilsocket

AceArd86, 2024-12-04:
I'm also having this same issue. It's in a boot loop

PR #876 (2021-04-18): Have watchdog plugin reboot system rather than restart application
The watchdog plugin detects the blindbug and restarts the application to fix the problem. Unfortunately its detection method relies on the kernel log changing so it doesn't constantly detect the blindbug. Rebooting the system instead of just restarting the application ensures that the kernel log is cleared.

Rebooting might also provide a better fix for the blindbug but this hasn't been tested.

Fixes #875

## Description
Replace call to pwnagotchi.restart with pwnagotchi.reboot.

## Motivation and Context

- [x] I have raised  [... cut]