evilsocket/pwnagotchi #878: Cannot run "sudo pip3 install -r requirements.txt" and "sudo pip3 install .".

heagan01, 2020-05-11:
**Describe the bug**
Cannot run "sudo pip3 install -r requirements.txt" and "sudo pip3 install .".

**To Reproduce**
Steps to reproduce the behavior:
1. cd pwnagotchi-1.4.3
2. sudo pip3 install -r requirements.txt
3. ...

**Expected behavior**
Installation completed

**Logs**
For the sudo pip3 install -r requirements.txt:
```
The directory '/home/heagan/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/heagan/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: crypto==1.4.1 in /usr/local/lib/python3.6/dist-packages (from -r requirements.txt (line 1))
Requirement already satisfied: requests==2.21.0 in /usr/local/lib/python3.6/dist-packages (from -r requirements.txt (line 2))
Requirement already satisfied: PyYAML==5.1 in /usr/local/lib/python3.6/dist-packages (from -r requirements.txt (line 3))
Requirement already satisfied: scapy==2.4.3 in /usr/local/lib/python3.6/dist-packages (from -r requirements.txt (line 4))
Requirement already satisfied: gym==0.14.0 in /usr/local/lib/python3.6/dist-packages (from -r requirements.txt (line 5))
Requirement already satisfied: scipy==1.3.1 in /usr/local/lib/python3.6/dist-packages (from -r requirements.txt (line 6))
Collecting stable-baselines==2.7.0 (from -r requirements.txt (line 7))
  Downloading https://files.pythonhosted.org/packages/52/9f/bbb2122cf2ce0e5f09a33bc35d504b83c05b0896bd46d4b85687c62557f6/stable_baselines-2.7.0-py3-none-any.whl (259kB)
    100% |████████████████████████████████| 266kB 89kB/s 
Collecting tensorflow==1.13.1 (from -r requirements.txt (line 8))
  Downloading https://files.pythonhosted.org/packages/77/63/a9fa76de8dffe7455304c4ed635be4aa9c0bacef6e0633d87d5f54530c5c/tensorflow-1.13.1-cp36-cp36m-manylinux1_x86_64.whl (92.5MB)
    100% |████████████████████████████████| 92.5MB 9.3kB/s 
Collecting tensorflow-estimator==1.14.0 (from -r requirements.txt (line 9))
  Downloading https://files.pythonhosted.org/packages/3c/d5/21860a5b11caf0678fbc8319341b0ae21a07156911132e0e71bffed0510d/tensorflow_estimator-1.14.0-py2.py3-none-any.whl (488kB)
    100% |████████████████████████████████| 491kB 1.1MB/s 
Collecting tweepy==3.7.0 (from -r requirements.txt (line 10))
  Downloading https://files.pythonhosted.org/packages/d5/5f/daac4b4e9b30d7d2a6fdd16a880ff79f27918fe388e4dfc1983dec3a9876/tweepy-3.7.0-py2.py3-none-any.whl
Collecting file-read-backwards==2.0.0 (from -r requirements.txt (line 11))
  Downloading https://files.pythonhosted.org/packages/53/cc/d015aacff2ae5566e81379b2146674c6bf364c50d562add9ae292a9284dc/file_read_backwards-2.0.0-py2.py3-none-any.whl
Requirement already satisfied: numpy==1.17.2 in /usr/local/lib/python3.6/dist-packages (from -r requirements.txt (line 12))
Collecting inky==0.0.5 (from -r requirements.txt (line 13))
  Downloading https://files.pythonhosted.org/packages/b8/29/0ce8c4bd259a5ac4e4657686d357ba7c45810adfcf6011ae84ac730bd1ad/inky-0.0.5-py3-none-any.whl
Collecting smbus2==0.3.0 (from -r requirements.txt (line 14))
  Downloading https://files.pythonhosted.org/packages/6a/06/80a6928e5cbfd40c77c08e06ae9975c2a50109586ce66435bd8166ce6bb3/smbus2-0.3.0.tar.gz
Collecting Pillow==5.4.1 (from -r requirements.txt (line 15))
  Downloading https://files.pythonhosted.org/packages/85/5e/e91792f198bbc5a0d7d3055ad552bc4062942d27eaf75c3e2783cf64eae5/Pillow-5.4.1-cp36-cp36m-manylinux1_x86_64.whl (2.0MB)
    100% |████████████████████████████████| 2.0MB 362kB/s 
Collecting spidev==3.4 (from -r requirements.txt (line 16))
  Downloading https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz
Collecting gast==0.2.2 (from -r requirements.txt (line 17))
  Downloading https://files.pythonhosted.org/packages/4e/35/11749bf99b2d4e3cceb4d55ca22590b0d7c2c62b9de38ac4a4a7f4687421/gast-0.2.2.tar.gz
Collecting flask==1.0.2 (from -r requirements.txt (line 18))
  Downloading https://files.pythonhosted.org/packages/7f/e7/08578774ed4536d3242b14dacb4696386634607af824ea997202cd0edb4b/Flask-1.0.2-py2.py3-none-any.whl (91kB)
    100% |████████████████████████████████| 92kB 2.5MB/s 
Collecting flask-cors==3.0.7 (from -r requirements.txt (line 19))
  Downloading https://files.pythonhosted.org/packages/65/cb/683f71ff8daa3aea0a5cbb276074de39f9ab66d3fbb8ad5efb5bb83e90d2/Flask_Cors-3.0.7-py2.py3-none-any.whl
Collecting flask-wtf==0.14.2 (from -r requirements.txt (line 20))
  Downloading https://files.pythonhosted.org/packages/60/3a/58c629472d10539ae5167dc7c1fecfa95dd7d0b7864623931e3776438a24/Flask_WTF-0.14.2-py2.py3-none-any.whl
Requirement already satisfied: shellescape in /usr/local/lib/python3.6/dist-packages (from crypto==1.4.1->-r requirements.txt (line 1))
Requirement already satisfied: Naked in /usr/local/lib/python3.6/dist-packages (from crypto==1.4.1->-r requirements.txt (line 1))
Requirement already satisfied: idna<2.9,>=2.5 in /usr/lib/python3/dist-packages (from requests==2.21.0->-r requirements.txt (line 2))
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/lib/python3/dist-packages (from requests==2.21.0->-r requirements.txt (line 2))
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests==2.21.0->-r requirements.txt (line 2))
Requirement already satisfied: urllib3<1.25,>=1.21.1 in /usr/lib/python3/dist-packages (from requests==2.21.0->-r requirements.txt (line 2))
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from gym==0.14.0->-r requirements.txt (line 5))
Requirement already satisfied: pyglet<=1.3.2,>=1.2.0 in /usr/local/lib/python3.6/dist-packages (from gym==0.14.0->-r requirements.txt (line 5))
Requirement already satisfied: cloudpickle~=1.2.0 in /usr/local/lib/python3.6/dist-packages (from gym==0.14.0->-r requirements.txt (line 5))
Collecting matplotlib (from stable-baselines==2.7.0->-r requirements.txt (line 7))
  Downloading https://files.pythonhosted.org/packages/93/4b/52da6b1523d5139d04e02d9e26ceda6146b48f2a4e5d2abfdf1c7bac8c40/matplotlib-3.2.1-cp36-cp36m-manylinux1_x86_64.whl (12.4MB)
    100% |████████████████████████████████| 12.4MB 70kB/s 
Requirement already satisfied: opencv-python in /usr/local/lib/python3.6/dist-packages (from stable-baselines==2.7.0->-r requirements.txt (line 7))
Collecting pandas (from stable-baselines==2.7.0->-r requirements.txt (line 7))
  Downloading https://files.pythonhosted.org/packages/bb/71/8f53bdbcbc67c912b888b40def255767e475402e9df64050019149b1a943/pandas-1.0.3-cp36-cp36m-manylinux1_x86_64.whl (10.0MB)
[... 466 lines cut ...]
    
    warning: build_ext: building optional extension "mpi4py.dl" failed
    
    checking for MPI compile and link ...
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c _configtest.c -o _configtest.o
    _configtest.c:2:10: fatal error: mpi.h: No such file or directory
     #include <mpi.h>
              ^~~~~~~
    compilation terminated.
    failure.
    removing: _configtest.c _configtest.o
    error: Cannot compile MPI programs. Check your configuration!!!
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-wu1i__in/mpi4py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-qt7rdmvy-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-wu1i__in/mpi4py/
```

**Environment (please complete the following information):**
 - Pwnagotchi 1.4.3
 - Linux Mint tricia
 - Intel Pentium
 - Zyrex sky 232x

**Additional context**
Sorry if I don't know anything about linux, I'm new here.

folletto95, 2022-02-03:
I have the same issue on an orangepi zero 2 running debian.
Some one how now a solution?

heagan01, 2022-02-05:
> I have the same issue on an orangepi zero 2 running debian. Some one how now a solution?

I've tried this on arch linux aswell it doesnt work

virtadpt, 2022-02-05:
Here's your problem:
```
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DHAVE_DLFCN_H=1 -DHAVE_DLOPEN=1 -I/usr/include/python3.6m -c src/dynload.c -o build/temp.linux-x86_64-3.6/src/dynload.o
     src/dynload.c:5:10: fatal error: Python.h: No such file or directory
     #include "Python.h"
```
You need to install the python3-dev package: `sudo apt-get install python3-dev`

folletto95, 2022-02-07:
> Here's your problem:
>
> ```
> x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC [...]

No change:

ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /tmp/tmp5fup82bm prepare_metadata_for_build_wheel /tmp/tmp5fy8dtei
         cwd: /tmp/pip-install-4jdjuxtb/scipy
    Complete output (137 lines):
    lapack_opt_info:
    lapack_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
      NOT AVAILABLE

    openblas_lapack_info:
    customize UnixCCompiler
    customize UnixCCompiler
      libraries openblas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
      NOT AVAILABLE

    openblas_clapack_info:
    customize UnixCCompiler
    customize UnixCCompiler
      libraries openblas,lapack not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
      NOT AVAILABLE

    atlas_3_10_threads_info:
    Setting PTATLAS=ATLAS
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/lib/aarch64-linux-gnu
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib/aarch64-linux-gnu
    <class 'numpy.distutils.system_info.atlas_3_10_threads_info'>
      NOT AVAILABLE

    atlas_3_10_info:
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/lib/aarch64-linux-gnu
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib/aarch64-linux-gnu
    <class 'numpy.distutils.system_info.atlas_3_10_info'>
      NOT AVAILABLE

    atlas_threads_info:
    Setting PTATLAS=ATLAS
    customize UnixCCompiler
      libraries ptf77blas,ptcblas,atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries ptf77blas,ptcblas,atlas not found in /usr/lib
[... 55 lines cut ...]
      self.calc_info()
    Traceback (most recent call last):
      File "/tmp/tmp5fup82bm", line 280, in <module>
        main()
      File "/tmp/tmp5fup82bm", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/tmp/tmp5fup82bm", line 133, in prepare_metadata_for_build_wheel
        return hook(metadata_directory, config_settings)
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 173, in prepare_metadata_for_build_wheel
        self.run_setup()
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 266, in run_setup
        super(_BuildMetaLegacyBackend,
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 157, in run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 505, in <module>
        setup_package()
      File "setup.py", line 501, in setup_package
        setup(**metadata)
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/numpy/distutils/core.py", line 135, in setup
        config = configuration()
      File "setup.py", line 403, in configuration
        raise NotFoundError(msg)
    numpy.distutils.system_info.NotFoundError: No lapack/blas resources found.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmp5fup82bm prepare_metadata_for_build_wheel /tmp/tmp5fy8dtei Check the logs for full command output.

folletto95, 2022-02-07:
> Here's your problem:
>
> ```
> x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC [...]

this is the resoult with python3-dev install

`root@orangepizero2:~/pwnagotchi-1.5.5#` sudo pip3 install -r requirements.txt
Collecting pycryptodome==3.9.4
  Using cached pycryptodome-3.9.4.tar.gz (11.6 MB)
Collecting requests==2.21.0
  Using cached requests-2.21.0-py2.py3-none-any.whl (57 kB)
Collecting PyYAML==5.3.1
  Using cached PyYAML-5.3.1.tar.gz (269 kB)
Collecting scapy==2.4.3
  Using cached scapy-2.4.3.tar.gz (905 kB)
Collecting gym==0.14.0
  Using cached gym-0.14.0.tar.gz (1.6 MB)
Collecting scipy==1.3.1
  Using cached scipy-1.3.1.tar.gz (23.6 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /tmp/tmp5fup82bm prepare_metadata_for_build_wheel /tmp/tmp5fy8dtei
         cwd: /tmp/pip-install-4jdjuxtb/scipy
    Complete output (137 lines):
    lapack_opt_info:
    lapack_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
      NOT AVAILABLE

    openblas_lapack_info:
    customize UnixCCompiler
    customize UnixCCompiler
      libraries openblas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
      NOT AVAILABLE

    openblas_clapack_info:
    customize UnixCCompiler
    customize UnixCCompiler
      libraries openblas,lapack not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
      NOT AVAILABLE

    atlas_3_10_threads_info:
    Setting PTATLAS=ATLAS
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/lib/aarch64-linux-gnu
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib/aarch64-linux-gnu
    <class 'numpy.distutils.system_info.atlas_3_10_threads_info'>
      NOT AVAILABLE

    atlas_3_10_info:
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/lib
    customize UnixCCompiler
[... 72 lines cut ...]
    Traceback (most recent call last):
      File "/tmp/tmp5fup82bm", line 280, in <module>
        main()
      File "/tmp/tmp5fup82bm", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/tmp/tmp5fup82bm", line 133, in prepare_metadata_for_build_wheel
        return hook(metadata_directory, config_settings)
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 173, in prepare_metadata_for_build_wheel
        self.run_setup()
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 266, in run_setup
        super(_BuildMetaLegacyBackend,
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 157, in run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 505, in <module>
        setup_package()
      File "setup.py", line 501, in setup_package
        setup(**metadata)
      File "/tmp/pip-build-env-t8evug7v/overlay/lib/python3.8/site-packages/numpy/distutils/core.py", line 135, in setup
        config = configuration()
      File "setup.py", line 403, in configuration
        raise NotFoundError(msg)
    numpy.distutils.system_info.NotFoundError: No lapack/blas resources found.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmp5fup82bm prepare_metadata_for_build_wheel /tmp/tmp5fy8dtei Check the logs for full command output.
`root@orangepizero2:~/pwnagotchi-1.5.5#`

heagan01, 2022-02-13:
it seems I got a building error with scipy?

```
Collecting crypto==1.4.1
  Using cached crypto-1.4.1-py2.py3-none-any.whl (18 kB)
Collecting requests==2.21.0
  Using cached requests-2.21.0-py2.py3-none-any.whl (57 kB)
Collecting PyYAML==5.1
  Using cached PyYAML-5.1.tar.gz (274 kB)
Collecting scapy==2.4.3
  Using cached scapy-2.4.3.tar.gz (905 kB)
Collecting gym==0.14.0
  Using cached gym-0.14.0.tar.gz (1.6 MB)
Collecting scipy==1.3.1
  Using cached scipy-1.3.1.tar.gz (23.6 MB)
  Installing build dependencies: started
  Installing build dependencies: still running...
  Installing build dependencies: finished with status 'error'
```