evilsocket/pwnagotchi #883: [BUG] Counter always 0 since 1.5.3

benallard, 2020-05-22:
**Describe the bug**
The total counter is increasing, but not the per-session counter. The SSID of the last pwnd network is not showing either.

**To Reproduce**
Steps to reproduce the behavior:
1. Start pwnagotchi
2. See the total counter increase
3. See the session counter stay at 0.

**Expected behavior**
The session counter should increase as well and show the last pwnd networks SSID.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.3

benallard, 2020-05-22:
[image]

This is in the logs.

LowSkillDeveloper, 2020-05-26:
The same thing, I created a new pwnagotchi and in the very first session I got 3(8). Several points were not displayed that they were found in the current session.