evilsocket/pwnagotchi #884: Setting up WiFi region in config.toml

LowSkillDeveloper, 2020-05-24:
By default, raspberry pi zero w in the raspbian image sets the region 00, World, in which limited channels from 1 to 11. From 12 to 14 are not available. In many countries channels 12-13 are used, but pwnagotchi does not work correctly with them. At least when I used it in manual mode. He does not see the point and cannot send packets at these frequencies, until I have use "iw reg set". Or did not change crda, but in this case, you need to change it every time during a flashing. And in the first case, use "iw reg set" each time.
And it will be more correct for people to set the region in which they are located.

LowSkillDeveloper, 2020-06-23:
If someone needs a workaround, then just add line "iw reg set BZ" to /etc/rc.local
Add before line "Exit".

Instead of BZ, you can specify any desired country. But BZ allows you to use 1-11 channels, as well as transmitter power up to 30.

Adapters that do not accept the country from the kernel will not be affected, here only modification crda.

Tuinslak, 2020-09-01:
Does that actually affect the RPi zeroW? `iw reg set XX` ?

LowSkillDeveloper, 2020-09-01:
> Does that actually affect the RPi zeroW? `iw reg set XX` ?

Yes. If you do not specify the region, then only channels from 1 to 11 are used. This does not affect the power. 12 and 13 are simply not used. And a lot of points are lost, at least in my region. As my country uses channels 1 to 13.

LowSkillDeveloper, 2020-09-01:
Start pwnagotchi, let it work in auto or ai mode and enter the "iw list" command in the console, and see which channels are enabled and which are not.

Or create a point on 13 channel and connect a device to it, pwnagotchi will not interact with it. And most likely he will not even notice her if she does not have an additional channel.

Tuinslak, 2020-09-02:
Cheers!

For whoever is reading this in the future — also worth noting that once you mess around with this, and you decide to keep it (ie by setting it in rc.local), you need to create a new brain or AI will error out.

So delete or rename the `brain.json` and `brain.nn` file in `/root/`.