evilsocket/pwnagotchi #887: [BUG] Changing bettercap credentials in config.toml breaks the pwnagotchi service

raspberrycoulis, 2020-06-06:
**Describe the bug**
Changing bettercap credentials in config.toml breaks the pwnagotchi service. The default username and password being `pwnagotchi` in both cases. Setting the username and password to anything else results in an authentication error when viewing the pwnagotchi logs.

**To Reproduce**
1. Flash fresh image (1.5.3) to SD card
2. Change `bettercap.username = "pwnagotchi"`and `bettercap.password = "pwnagotchi"` to anything you like
3. Reboot
4. pwnagotchi goes straight into AI mode and does not work.

**Expected behavior**
It should work.

**Screenshots**
Not needed.

**Environment (please complete the following information):**
 - Pwnagotchi version - 1.5.3
 - OS version - Raspbian Lite (as per the image that comes with 1.5.3)
 - Pi0 W but also on any other Pi model including 4

xenDE, 2020-06-07:
if you want to change the password, you need to change it in bettercap too/first.

in the pwnagotchi config you only define the password for use bettercap.

that's not a bug!

raspberrycoulis, 2020-06-07:
> if you want to change the password, you need to change it in bettercap too/first.
> 
> in the pwnagotchi config you only define the password for use bettercap.
> 
> that's not a bug!

Ah. I was confused then because on the wiki it talks about changing the credentials for the pwnagotchi UI in the .TOML file there, so I assumed you do the same for bettercap too.

fennectech, 2020-10-12:
where do you set the password in bettercap

closed 2020-10-21 by raspberrycoulis