evilsocket/pwnagotchi #894: Support for Adafruit PiTFT

nachtschatt3n, 2020-06-26:
It would be great if Pwnagotchi would support the Adafruit PiTFT LCD Family. The console output drivers are supplied by Adafruit it should be easy to stitch it together. 

https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi

agentages, 2020-10-15:
https://pwnagotchi.ai/community/#waveshare-3-5-spi-tft-screen

It says this method should work with other framebuffer based displays, give it a try.

texastakk, 2021-01-07:
I've seen photos and, I think, a video of it working, about to give it a go since the waveshares keep dying on me. I think I have a 2.2 tft and a 3.5 pybadge or something. Not sure how aggravating it will be to configure so I'm accepting all advice lol

Crash-Log, 2021-11-14:
I came across a solution for this, I basically installed the Pitft drivers with the adafruit script on the pwnagotchi ready to use image, then followed this :
https://github.com/evilsocket/pwnagotchi/issues/124
starting from the "Reboot" step and used "spotpear24inch" in the display type instead of "inky"

texastakk, 2021-11-15:
Hey thanks for answering I'll have to try it soon