evilsocket/pwnagotchi #896: [BUG] GPS module not loaded

CptanPanic, 2020-07-22:
**Describe the bug**
Connected a USB GPS Module, and it shows up as /dev/ttyACM0, but the logs say "error 400: module not loaded"

**To Reproduce**
Steps to reproduce the behavior:
1. Connected Ublox7 USB serial device to microusb connector.

**Expected behavior**
GPS Connection, and shown on screen.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.3
 - Type of hardware Pi Zero W+

virtadpt, 2020-07-23:
Can confirm.  At first, I thought it was because I was running gpsd to interact with the GPS (to also gather Openstreetmap data, and set the internal clock correctly) but the same thing happens when it's not running or installed (I upgraded Dalton to image v1.5.3 recently).

crahan, 2020-08-13:
Running into the same issue with a GlobalSat 4 device on /dev/ttyUSB0 (baudrate 4800). I can read data from the device but the logs show the "error 400: module not loaded" error and no coordinates are displayed.

Hexalt, 2020-08-23:
I had a similar issue using a L76X gps module on GPIO pins. Using /dev/serial0 failed but /dev/ttyS0 seemed to work.

see if /dev/ttyACM0 is linked to another dev. If it is try using that instead.
`ls -lah /dev/`

fishgrind, 2020-09-02:
Hey, this might help you.

I had the same issue, worked fine in BetterCap ui if I set the gps to dev/rfcomm1 (i am using a bt gps mouse)

Didn't have any data in pwnagotchi though.

What i did was edit /usr/local/share/bettercap/caplets/gps.cap and changed it to dev/rfcomm1 and it instantly started to work on the pwnagotchi.

Retsim, 2020-09-17:
Hey,

Had the exact same issue.
I finally created a new plugin "unoficialgps.py", see the pwnagotchi slack for my explanations.
There: https://pwnagotchi.slack.com/archives/CNHCTKFJP/p1600349293099900

My guess is the link between pwnagotchi and bettercap was not working well in my case.
My GPS module was working fine through command line...

My unoficialgps.py should have the same behavior as the default gps plugin, but instead of relying on bettercap, it directly reads the /dev/ttyACM0 (or whatever) by itself using a Thread.

crahan, 2020-09-17:
I just realized that my issue was a PEBKAC situation. After reviewing the code I noticed that the `error 400: module not loaded` comes from the initial `agent.run("gps off")` command that is sent prior to configuring the Bettercap GPS settings and turning the GPS on via `agent.run("gps on")` further down in the code.

When `agent.run("gps off")` is called (most likely to make sure that the plugin is not loaded twice in Bettercap) the agent can't find a running GPS plugin and thus throws the error. So it seems to be (at least in my case) expected behavior. As for not seeing any data, I mistakingly thought that the Pwnagotchi would show the current GPS coordinates at certain intervals, but instead it will show the GPS coordinates for each successful handshake it captures.

After waiting for my device to capture a handshake I saw the coordinates appear on screen. Just thought I'd share.

adi170-alt, 2023-01-11:
Is there already fix for then?(with using the stock GPS plugin)