evilsocket/pwnagotchi #923: [BUG] Please update APT signing key for http.re4son-kernel.com

Daviey, 2020-09-26:
**Describe the bug**
APT signing key for http.re4son-kernel.com has expired and been updated with a new expiry.

**To Reproduce**

Can be demonstrated with an `apt update`

```
$ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                                                            
Hit:2 http://archive.raspberrypi.org/debian buster InRelease                                                                                                               
Get:3 http://http.re4son-kernel.com/re4son kali-pi InRelease [8,133 B]                                                            
Get:4 https://pkgs.tailscale.com/stable/raspbian buster InRelease                
Err:3 http://http.re4son-kernel.com/re4son kali-pi InRelease                                                                                                                                              
  The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving <carsten.boeving@whitedome.com.au>
Fetched 11.6 kB in 13s (928 B/s)                                                                                                                                                                          
Reading package lists... Done
Building dependency tree       
Reading state information... Done
82 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://http.re4son-kernel.com/re4son kali-pi InRelease: The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving <carsten.boeving@whitedome.com.au>
W: Failed to fetch http://http.re4son-kernel.com/re4son/dists/kali-pi/InRelease  The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving <carsten.boeving@whitedome.com.au>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

**Solution**

Either:

`sudo apt-key adv --keyserver keys.gnupg.net --recv-keys 11764EE8AC24832F`
`wget -O - https://re4son-kernel.com/keys/http/archive-key.asc | sudo apt-key add -`

(But please include this key in the next release, kkthnxbye)

Motophan, 2021-04-04:
any updates on this?

marko-kovacevic, 2021-04-15:
I'm still having this issue - downloaded the image yesterday

Motophan, 2021-04-15:
Any update on this?

lolzdatguy, 2021-04-16:
I'm pretty sure evilsocket is done with this project, check out Dadav's fork.

r12w4n, 2021-05-23:
# nano /etc/apt/sources.list
deb http://ftp.harukasan.org/kali kali-rolling main contrib non-free

qkum, 2022-02-09:
> 

Non-Https download link? 

Is this a joke?

I don't know if you are the owner but never link something that is not https for god sake....

Unless the intention is to get people Hacked?

qkum, 2022-02-09:
> 

None of these fixes work February 2022 it seems.

qkum, 2022-02-09:
The solution seems to be one of the many ones provided by a lot of lovely people in this threat: `https://unix.stackexchange.com/questions/421821/invalid-signature-for-kali-linux-repositories-the-following-signatures-were-i`

Daviey, 2022-02-09:
> > 
> 
> Non-Https download link?
> 
> Is this a joke?
> 
> I don't know if you are the owner but never link something that is not https for god sake....
> 
> Unless the intention is to get people Hacked?

What is the threat you are trying to mitigate?  If it is confidentiality/privacy then you are entirely correct, however if it is integrity or some other threat i'd like to understand your concern more.  Can you expand please?  Thanks

qkum, 2022-02-09:
> What is the threat you are trying to mitigate? If it is confidentiality/privacy then you are entirely correct, however if it is integrity or some other threat i'd like to understand your concern [...]

I'm not a pro hacker and I don't want to be. 

But I know that if the connection is not encrypted in theory the file can be swapped out and I get hacked.

We did not invent https to not use it. (every legit website use it for a reason)

Thanks for your curious question though. At least you seem to care.

Daviey, 2022-02-09:
> > > > 
> > > 
> > > 
> > > Non-Https download link?
> > > Is this a joke?
> > > I don't know if you are the owner but never link something that is not https for god sake....
> > > Unless the intention is to get people Hacked?
> > 
> > 
> > What is the threat you are trying to mitigate? If it is confidentiality/privacy then you are entirely correct, however if it is integrity or some other threat i'd like to understand your concern more. Can you expand please? Thanks
> 
> I'm not a pro hacker and I don't want to be.
> 
> But I know that if the connection is not encrypted in theory the file can be swapped out.
> 
> We did not invent https to not use it. (every legit website use it for a reason)
> 
> Thanks for your curious question though. At least you seem to care.

Ah, the APT archive is signed using GPG and the integrity of each of the packages is assured by verifying the hash of the packages against a GPG signed index.  Therefore it isn't possible for it to be swapped out by a malicious actor.

However, if you are concerned someone is listening to your traffic between yourself and the APT mirror, they could see what packages you are installing which for a security centric project could put you at risk.  However, the contents cannot be modified.

(This issue was specifically about the GPG key expiring, which is used to verifying the signed index.)

Anyway, glad you were able to make progress!  I have no idea if this is still an issue on a clean install of pwnagotchi.

PyMarc2, 2022-09-12:
No solution works as in September 2022.

st33ve0, 2022-09-15:
> No solution works as in September 2022.

Try this:

sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F

https://twitter.com/re4sonkernel/status/1299856596256784385?lang=en

Ra2-IFV, 2022-09-20:
> sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F

Worked.

PyMarc2, 2022-09-20:
```
└─$ sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F
[sudo] password for pythagoras:
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.t6lhcV9Z31/gpg.1.sh --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F
gpg: key 11764EE8AC24832F: "Carsten Boeving <carsten.boeving@whitedome.com.au>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

┌──(pythagoras㉿alicesys)-[~]
└─$ sudo apt-get update
Hit:2 http://repo.mysql.com/apt/debian bullseye InRelease
Ign:4 https://packages.sury.org/php kali-rolling InRelease
Err:5 https://packages.sury.org/php kali-rolling Release
  404  Not Found [IP: 138.199.57.151 443]
Hit:1 http://kali.download/kali kali-rolling InRelease
Hit:3 http://kali.download/kali kali-experimental InRelease
Hit:6 http://http.re4son-kernel.com/re4son kali-pi InRelease
Reading package lists... Done
E: The repository 'https://packages.sury.org/php kali-rolling Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: http://repo.mysql.com/apt/debian/dists/bullseye/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
N: Skipping acquire of configured file 'mysql-apt-config/binary-arm64/Packages' as repository 'http://repo.mysql.com/apt/debian bullseye InRelease' doesn't support architecture 'arm64'
N: Skipping acquire of configured file 'mysql-8.0/binary-arm64/Packages' as repository 'http://repo.mysql.com/apt/debian bullseye InRelease' doesn't support architecture 'arm64'
N: Skipping acquire of configured file 'mysql-tools/binary-arm64/Packages' as repository 'http://repo.mysql.com/apt/debian bullseye InRelease' doesn't support architecture 'arm64'
```

Ra2-IFV, 2022-09-22:
> Worked for the invalid error, but Instead of invalid, now it is not found.
Are you sure you are using the official image? I didn't see these repos such as `http://kali.download/kali` in my newly flashed pwnagotchi.

Crazy-Tux, 2022-09-25:
> > No solution works as in September 2022.
> 
> Try this:
> 
> sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F
> 
> https://twitter.com/re4sonkernel/status/1299856596256784385?lang=en

This solution is working well.

MaffooClock, 2022-10-24:
I tried using `sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F`, but kept getting "gpg: keyserver receive failed: Server indicated a failure."  Also tried port 80, same problem.

I'm sure whatever the problem is must be on my end, but I ended up doing it manually:
1.  Copy/paste the key block from https://pgp.mit.edu/pks/lookup?op=get&search=0x11764EE8AC24832F into a new file, `11764EE8AC24832F.asc` (wget won't work because you get an HTML document)
2. Execute `sudo apt-key add 11764EE8AC24832F.asc`

cielobao, 2022-11-17:
Two years later, the issues persists. 

"sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F"

returns

gpg: keyserver received failed: No keyserver available

Manual copy and paste followed by sudo apt-key add failed.
Who has the updates?

Console, 2022-11-18:
@Cielobao that solution simply adds a bunch of kali stuff to your sources list. Some of those packages will not be supported by your kernel.

Instead visit a keyserver directly like http://keyserver.ubuntu.com and search for the key ID you need. e.g. in this case...

https://keyserver.ubuntu.com/pks/lookup?search=0x11764EE8AC24832F&fingerprint=on&op=index

Which will return the stored key entry. (the issue is that the dirmmgr (a utility called by apt-keys) on the pwnagotchi build has a known bug with keyservers (https://dev.gnupg.org/T3374)

Then click the link for the public key on that page or alternatively just click here: https://keyserver.ubuntu.com/pks/lookup?op=get&search=0x96d00b0c4b00fd5360cc087711764ee8ac24832f

Copy and paste that into a file and use the command:
```cat file.ext | sudo apt-key add -```

You'll get this...

[image]

HugoDL, 2023-01-08:
This one worked here:

`sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F`

NotaRonin, 2023-02-15:
> @cielobao that solution simply adds a bunch of kali stuff to your sources list. Some of those packages will not be supported by your kernel.
>
> Instead visit a keyserver directly like [...]
Worked like a charm, just had to run:
sudo apt update && sudo apt full-upgrade -y 
afterwards to deal with broken packages.

awnyel, 2023-05-26:
> This one worked here:
> 
> `sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F`

this works perfect thx!

cdkw, 2023-06-23:
I have copied everything (ctrl+a) from [http://keyserver.ubuntu.com/pks/lookup?op=get&search=0xed444ff07d8d0bf6](http://keyserver.ubuntu.com/pks/lookup?op=get&search=0xed444ff07d8d0bf6) and made a new file named `ED444FF07D8D0BF6.asc` with its contents.
Then did `cat ED444FF07D8D0BF6.asc | sudo apt-key add -`
It returned `OK` and now works perfectly.

ofsanjay1, 2023-07-06:
Worked for me also Thanks 
**Solution**

 _sudo apt-key adv --keyserver hkp://pgp.mit.edu --recv-keys 11764EE8AC24832F_

**ERROR**
Reading state information... Done
322 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://http.re4son-kernel.com/re4son kali-pi InRelease: The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving [carsten.boeving@whitedome.com.au](mailto:carsten.boeving@whitedome.com.au)
W: Failed to fetch http://http.re4son-kernel.com/re4son/dists/kali-pi/InRelease  The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving [carsten.boeving@whitedome.com.au](mailto:carsten.boeving@whitedome.com.au)
W: Some index files failed to download. They have been ignored, or old ones used instead.

BENDER35, 2023-07-11:
I have tried with this command

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 11764EE8AC24832F

and update the re4son packages

[image]

greetings

fluid46, 2023-09-08:
`wget -q -O - http://http.re4son-kernel.com/re4son/kali_pi-archive-keyring.gpg | apt-key add`

-Worked for me

joneschrisan, 2024-03-03:
> `wget -q -O - http://http.re4son-kernel.com/re4son/kali_pi-archive-keyring.gpg | apt-key add`
> 
> -Worked for me

missing th '-' on the end and apt-key has to be ran as root, says 'OK' after adding the key but still shows invalid key on apt update

`$ wget -q -O - http://http.re4son-kernel.com/re4son/kali_pi-archive-keyring.gpg | sudo apt-key add -
OK
$ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Get:3 http://http.re4son-kernel.com/re4son kali-pi InRelease [10.4 kB]
Err:3 http://http.re4son-kernel.com/re4son kali-pi InRelease
  The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving <carsten.boeving@whitedome.com.au>
Fetched 10.4 kB in 3s (3,203 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
256 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://http.re4son-kernel.com/re4son kali-pi InRelease: The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving <carsten.boeving@whitedome.com.au>
W: Failed to fetch http://http.re4son-kernel.com/re4son/dists/kali-pi/InRelease  The following signatures were invalid: EXPKEYSIG 11764EE8AC24832F Carsten Boeving <carsten.boeving@whitedome.com.au>
W: Some index files failed to download. They have been ignored, or old ones used instead.`

However usin the link to the actual hosted file in the repo `http://http.re4son-kernel.com.s3.amazonaws.com/re4son/kali_pi-archive-keyring.gpg` gives:
`$ wget -q -o http://http.re4son-kernel.com.s3.amazonaws.com/re4son/kali_pi-archive-keyring.gpg | sudo apt-key add -
wget: http://http.re4son-kernel.com.s3.amazonaws.com/re4son/kali_pi-archive-keyring.gpg: No such file or directory
gpg: no valid OpenPGP data found.`

Could this be the reason for the error

Shao-Chen-Zhang, 2024-12-27:
If anyone is still having this issue, I managed to fix this on my Raspberry Pi 4 by running the following command:
`wget -O /etc/apt/trusted.gpg.d/kali_pi-archive-keyring.gpg https://re4son-kernel.com/keys/http/kali_pi-archive-keyring.gpg`

This is straight from the build script in the Kali Linux repository which you can find here:
https://gitlab.com/kalilinux/build-scripts/kali-arm/-/blob/main/raspberry-pi.sh?ref_type=heads