evilsocket/pwnagotchi #924: Is it possible to restore wlan0?

subokita, 2020-09-29:
Is it possible to restore wlan0?

The reason why I'm asking, because my Mac Catalina can't connect to pwnagotchi USB port no more (no RNDIS/Ethernet Gadget showing up, tried multiple methods, doesn't work, and with Big Sur coming out soon, no one is sure if HORNDIS would still work, ifconfig on pwnagotchi does show usb0 on 10.0.0.2)

Thus, now I have a pwngotchi that works, but I can only read the data by using HDMI, rather than extracting it from my main computer, thru, say SSH, FTP, or the web UI.

What's the step to restore wlan0?
1. ifconfig mon0 down
2. ifconfig wlan0 up?
3. set the wifi access point thru either wpa_supplicant.conf or raspi-config? Which doesn't seem to work.

forceshield, 2020-10-03:
'sudo /usr/bin/monstop' will turn off the monitor mode.  

I've had a bit of trouble trying to get mine connected to wifi after that, raspi-config gives me wpa_supplicant errors(could not communicate), and iwlist scan on gives me network not available.  

If you're more familiar than me with connecting via terminal you may have better luck catching something I'm missing - let me know if you do please

debian-user-france1, 2020-11-13:
please close this issue

closed 2020-11-13 by subokita