evilsocket/pwnagotchi #925: [BUG] AI training on Pi3b+

quangvu1991, 2020-09-29:
**Describe the bug**
pwnagotchi work fine in auto mode. when switch from auto to AI mode, it got error in logs and not discover AP anymore. 

logs:
[2020-09-28 14:31:02,190] [ERROR] [ai] error while training (could not broadcast input array from shape (428) into shape (1,503))
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 231, in learn
    runner = A2CRunner(self.env, self, n_steps=self.n_steps, gamma=self.gamma)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 310, in __init__
    super(A2CRunner, self).__init__(env=env, model=model, n_steps=n_steps)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/runners.py", line 19, in __init__
    self.obs[:] = env.reset()
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 52, in reset
    self._save_obs(env_idx, obs)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 71, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (428) into shape (1,503)

**To Reproduce**
Steps to reproduce the behavior:
1. fresh install pwnagotchi v1.5.3 on raspberry pi3b+
2. wait for pwnagotchi switch from auto to AI mode.

**Expected behavior**
pwnagotchi continue find AP, handshake in AI mod

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.5.3
 - OS version: pwnagotchi 1.5.3 img
 - Type of hardware: Raspberry Pi 3b+
- Another hardware: waveshares 3.5 screen (B)

debian-user-france1, 2020-11-08:
I don't have this issue... I'm using also a Rasperrypi 3 B +...

But if it works a long time and restart much times, it wont switch to AI anymore....

your problem may in the configuration file of pwnagotchi, in /etc/pwnagotchi/

guycalledseven, 2020-12-31:
I have the same issue except I am not getting any APs at all. :-(
I managed to "fix" python AI issue by loading bunch of dependencies:

```sudo apt-get install libopencv-dev python-opencv python-picamera```

Hope it helps.

debian-user-france1, 2021-01-02:
I've the bug, that the pwnagotchi won't switch to the ai mode. It log files say, that the error occurred while reading the brain.nn configuration file. I fixed it with removing the file. I don't know, if it is this file or the other file. But it is the "model"

Does that affect to the pwnagotchi?
The ai worked perfectly after restarting the pwnagotchi

pastel-parasol, 2021-01-03:
Same issue.
Tried @guycalledseven 's solution, but nothing changed.
Using Raspberry Pi 3A+, same error in logs.
Also, I can't access bettercap's interface in auto/manual mode.

Saru13, 2021-05-23:
Tried 1.5.5?
Can we close this?

facastagnini, 2022-02-11:
I had exactly the same error reported here https://github.com/evilsocket/pwnagotchi/issues/925#issue-711194618

### The error

> 16:59:09 [INFO] [ai] model created:
> 16:59:09 [INFO]       gamma: 0.99
> 16:59:09 [INFO]       n_steps: 1
> 16:59:09 [INFO]       vf_coef: 0.25
> 16:59:09 [INFO]       ent_coef: 0.01
> 16:59:09 [INFO]       max_grad_norm: 0.5
> 16:59:09 [INFO]       learning_rate: 0.001
> 16:59:09 [INFO]       alpha: 0.99
> 16:59:09 [INFO]       epsilon: 1e-05
> 16:59:09 [INFO]       verbose: 1
> 16:59:09 [INFO]       lr_schedule: constant
> 16:59:10 [INFO] [ai] learning for 50 epochs ...
> 16:59:10 [ERROR] [ai] error while training (could not broadcast input array from shape (428,) into shape (1,503))
> (most recent call last):
>  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
>    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
>  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 231, in learn
>    runner = A2CRunner(self.env, self, n_steps=self.n_steps, gamma=self.gamma)
>  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 310, in __init__
>    super(A2CRunner, self).__init__(env=env, model=model, n_steps=n_steps)
>  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/runners.py", line 19, in __init__
>    self.obs[:] = env.reset()
>  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 52, in reset
>    self._save_obs(env_idx, obs)
>  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 71, in _save_obs
>    self.buf_obs[key][env_idx] = obs
> could not broadcast input array from shape (428,) into shape (1,503)

### Hardware
`
$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : XXXXXXXXXXXXXXX
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
`

### Tried 2 clean installations
- 1.5.3 official image
  - flashed the image
  - plugged the rpi and let it do its thing for a while. It automatically updated pwnagotchi to 1.5.5 and rebooted.
  - Eventually the AI engages and the error occurs.
- 1.5.5 official image
  - flashed the image
  - plugged the rpi and let it do its thing for a while.
  - sudo pip3 install --upgrade numpy (to fix a known issue, numpy-1.21.5 installed)
  - rebooted and waited
  - Eventually the AI engages and the error occurs.
  

In both instances, there is no _/boot/brains.nn_ being created.

Thanks in advance for your help

tomasjanu, 2022-04-12:
I have the same issue as [thebitsworld](https://github.com/thebitsworld) and [facastagnini](https://github.com/facastagnini).

Same error, no brain.nn created :(

tomasjanu, 2022-04-13:
Interesting observation!

Taking the microSD card from RPi 3B+ to Rpi Zero (so I didnt change anything else) and the AI works perfectly!

So now... let's find out what the AI actually does :D

tomasjanu, 2022-04-13:
> I have the same issue as [thebitsworld](https://github.com/thebitsworld) and [facastagnini](https://github.com/facastagnini).
> 
> Same error, no brain.nn created :(

Interesting observation!

Taking the microSD card from RPi 3B+ to Rpi Zero (so I didnt change anything else) and the AI works perfectly!

So now... let's find out what the AI actually does :D

XZCendence, 2022-06-09:
Having this same issue :( Not sure how to fix it