evilsocket/pwnagotchi #930: Help

LamonJ, 2020-10-06:
I am kinda new to this but is there somewhere I can go to figure out how to install plug-ins or any other .py variation? For instance, how do you tell your pwnagotchi to look for those other .py scripts? Like I said I am kinda new to this and want to learn more.

thecoder-001, 2020-10-07:
The pwnagotchi website has a page on many available plugins (both default and community made) and how to apply them [here](https://pwnagotchi.ai/plugins/). For default plugins, you just need to enable them in your `config.toml` and add the necessary parameters (You can find out all the individual plugin parameters by referring to [default.toml](https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/defaults.toml)).
For community made plugins, you need to first download them from the internet and then enable them in the `config.toml`.
You can find some of the most used community plugins [here](https://github.com/evilsocket/pwnagotchi-plugins-contrib) in evilsocket's repo. However, the config files there have not yet been updated to the new `toml` format (and is in the old yml format), so you can refer to my fork [here](https://github.com/thecoder-001/pwnagotchi-plugins-contrib) for the new format and instructions. Also, keep in mind that the community plugins are not been completely tested by the dev team, use them at your own risk.
For more community plugins, you can search on the internet with keywords like "pwnagotchi" + "community" + "plugins".

appu1282, 2020-10-07:
[image]_

Guys I am getting this issue so someone can - help me please _______

thecoder-001, 2020-10-07:
Hello @appu1282 . Your issue doesn't seems to be related to this issue thread. Please open a new issue for help (if its related to pwnagotchi if not please open a issue in the respective project's repo). You seem to be trying to install Kali Linux tools on a local Termux environment (I'm sorry if I'm wrong). Installing pwnagotchi on a android phone is not impossible but is certainly hard. You would need custom kernel to allow monitor mode (if your wifi card supports). Still, for a head start the error indicates that either python is not installed on your system or its not correctly placed. You can try `sudo apt install python3.8` to install python.
From the next time, please try to open a new issue if its not relevant to others.
Thanks.

LamonJ, 2020-10-10:
Thank you for your feedback, I will check those resources.