evilsocket/pwnagotchi #931: Please help this is multiple weeks worth of frustration

Smurfy37, 2020-10-08:
ive installed rasbian on my raspberry pi i intend on running bettercap locally and using the pixel desktop to run chromium in full screen with the face open but every time i try to launch with sudo pwnagotchi --manual it gives me this error and i cant find diddly squat i feel very stupid please save me

pi@Tony:~ $ sudo pwnagotchi --manual
 [INFO] [update] plugin loaded.
 [INFO] grid plugin loaded.
 [INFO] Session-stats plugin loaded.
 [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
 [WARNING] ui.fps is 0, the display will only update for major changes
 [INFO] initializing waveshare v2 display
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 131, in <module>
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 72, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/waveshare2.py", line 58, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.v2.waveshare import EPD
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py", line 60, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory
pi@Tony:~ $

i find it especially odd that i dont even have an Eink display attached so it attempting to use a waveshare screen will obviously fail

derconno, 2020-10-08:
can you post your /etc/pwnagotchi/config.toml?

Smurfy37, 2020-10-09:
main.name = "Tony"
main.lang = "en"
main.whitelist = [
  "EXAMPLE_NETWORK",
  "ANOTHER_EXAMPLE_NETWORK",
  "fo:od:ba:be:fo:od",
  "fo:od:ba"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "YourHomeNetworkHere"
]

ui.display.enabled = false
ui.display.type = "inky"
ui.display.color = "black"
----------------------------------------------------

After changing to false i can confirm that it gives me the same error

Smurfy37, 2020-10-09:
im open to any suggestions

Smurfy37, 2020-10-09:
ok after some fiddling ive managed to change the error its been trying to go to config.toml and when i checked /usr/local/bin/pwnagotchi    i found  its not looking for that config at all its looking for config.yml so i did a cp to make it that and suddenly its giving me this error ive tried setting ui.display.type to inky and display but neither work giving me the following error

pi@Tony:~ $ sudo pwnagotchi --manual
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 120, in <module>
    config = utils.load_config(args)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/utils.py", line 84, in load_config
    if config['ui']['display']['type'] in ('inky', 'inkyphat'):
TypeError: string indices must be integers

Smurfy37, 2020-10-09:
the type error would suggest that im not supposed to be entering text into that bit of the config but what numbers would i use as substitute

derconno, 2020-10-09:
okay, it seems you are using an older version of "pwnagotchi" that relies on .yml config files. Please try to install the current version from [here](https://github.com/evilsocket/pwnagotchi/releases/tag/v1.5.3)

Smurfy37, 2020-10-09:
thankyou ima try this now 👍

Smurfy37, 2020-10-09:
pi@Tony:~ $ sudo bettercap -eval "caplets.update; ui.update; quit"
 _          _   _
| |__   ___| |_| |_ ___ _ __ ___ __ _ _ __
| '_ \ / _ \ __| __/ _ \ '__/ __/ _` | '_ \
| |_) |  __/ |_| ||  __/ | | (_| (_| | |_) |
|_.__/ \___|\__|\__\___|_|  \___\__,_| .__/
                                     |_| v1.6.2
http://bettercap.org/

[E] 'Invalid option: -eval', verify your command line arguments executing 'bettercap --help'.

any ideas?

Smurfy37, 2020-10-09:
ive just realised you dont actually have a link to the pwnagotchi v1.5.3 by itself without raspbian

Smurfy37, 2020-10-09:
this link is outdated and changing the numbers just downloads a html unless im missing something please call me an idiot if its due XD
[image]

derconno, 2020-10-09:
the default raspbian image should work for any pi, so using it could be an option. Also, [this url](https://github.com/evilsocket/pwnagotchi/archive/v1.5.3.zip) can be used to download the current version

Smurfy37, 2020-10-09:
thanks man i got it working im just stuck on the bettercap working properly now its just constantly waiting for bettercap api to be avaliable

Smurfy37, 2020-10-09:
i just get this endlessly scrolling

[2020-10-09 21:19:51,861] [INFO] error 404: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN">
<HTML>
  <HEAD><TITLE>Not Found</TITLE></HEAD>
  <BODY>
    <H1>Not Found</H1>
    `/api/session' not found.
    <HR>
    <ADDRESS>
     WEBrick/1.4.2 (Ruby/2.5.5/2019-03-15) at
     localhost:8081
    </ADDRESS>
  </BODY>
</HTML>
[2020-10-09 21:19:51,862] [INFO] waiting for bettercap API to be available ...
[2020-10-09 21:19:53,337] [INFO] error 404: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN">
<HTML>
  <HEAD><TITLE>Not Found</TITLE></HEAD>
  <BODY>
    <H1>Not Found</H1>
    `/api/session' not found.
    <HR>
    <ADDRESS>
     WEBrick/1.4.2 (Ruby/2.5.5/2019-03-15) at
     localhost:8081
    </ADDRESS>
  </BODY>
</HTML>
[2020-10-09 21:19:53,338] [INFO] waiting for bettercap API to be available ...
[2020-10-09 21:19:54,347] [INFO] error 404: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN">
<HTML>
  <HEAD><TITLE>Not Found</TITLE></HEAD>
  <BODY>
    <H1>Not Found</H1>
    `/api/session' not found.
    <HR>
    <ADDRESS>
     WEBrick/1.4.2 (Ruby/2.5.5/2019-03-15) at
     localhost:8081
    </ADDRESS>
  </BODY>
</HTML>
[2020-10-09 21:19:54,347] [INFO] waiting for bettercap API to be available ...
[2020-10-09 21:19:55,362] [INFO] error 404: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN">
<HTML>
  <HEAD><TITLE>Not Found</TITLE></HEAD>
  <BODY>
    <H1>Not Found</H1>
    `/api/session' not found.
    <HR>
    <ADDRESS>
     WEBrick/1.4.2 (Ruby/2.5.5/2019-03-15) at
     localhost:8081
    </ADDRESS>
  </BODY>
</HTML>
[2020-10-09 21:19:55,363] [INFO] waiting for bettercap API to be available ...
[2020-10-09 21:19:56,488] [INFO] error 404: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN">
<HTML>
  <HEAD><TITLE>Not Found</TITLE></HEAD>
  <BODY>
    <H1>Not Found</H1>
    `/api/session' not found.
    <HR>
    <ADDRESS>
     WEBrick/1.4.2 (Ruby/2.5.5/2019-03-15) at
     localhost:8081
    </ADDRESS>
  </BODY>
</HTML>
[2020-10-09 21:19:56,489] [INFO] waiting for bettercap API to be available ...
[2020-10-09 21:19:57,649] [INFO] error 404: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN">
<HTML>
  <HEAD><TITLE>Not Found</TITLE></HEAD>
  <BODY>
    <H1>Not Found</H1>
    `/api/session' not found.
    <HR>
    <ADDRESS>
     WEBrick/1.4.2 (Ruby/2.5.5/2019-03-15) at
     localhost:8081
    </ADDRESS>
  </BODY>
</HTML>
[2020-10-09 21:19:57,650] [INFO] waiting for bettercap API to be available ...

agentages, 2020-10-10:
Have you not just flashed the pwnagotchi image to see if that works for you?

Smurfy37, 2020-10-10:
ive tried that twice and it never worked i just had a blinking _ in the top corner i left it for an hour and it didnt move and ive got a pi4 i feel like im getting close tho i just need to figure out what it means by its not found so bettercap works with pwnagotchi

agentages, 2020-10-13:
Did you change any passwords or ports in your config?

Smurfy37, 2020-10-16:
i have not changed the port, the top command doesnt run and the config is the same as im using on my manual pi desktop install ive also wondered if its because im using an led display rather than an E-ink display

forceshield, 2020-10-18:
Get into a root shell with sudo -i
then try the command
lsof -i :8081

I'm also using an LCD display (240x240 LCDhat)

debian-user-france1, 2020-11-13:
I would do this:
try again flashing with newest version, CHECK SHASUM! 
or
buy new sd card and try again, I have two raspberry pis, one 3B and one 3B+