evilsocket/pwnagotchi #934: Config.toml

LamonJ, 2020-10-11:
Where in the hell do I find the Config.toml file? I downloaded the master and I was looking through it and I can not find it anywhere.
I have gone through the documentation on the pwnagotchi.ai site and can not find the information I was needing.

thecoder-001, 2020-10-11:
It's named as `defaults.toml` and is placed in here `pwnagotchi/`[defaults.toml](https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/defaults.toml) in the repo.
It has also been mentioned in the [pwnagotchi.ai](www.pwnagotchi.ai) documentation website [here](https://pwnagotchi.ai/configuration/#initial-configuration).

LamonJ, 2020-10-12:
as I understood it, you should not mess with the default.toml. Also, I understood it that you had to save "your" config.toml to the root of the SD card and that when you first boot up it will save your changes to a config.toml somewhere. Am I understanding that correctly?

thecoder-001, 2020-10-13:
Yes, you have understood it all correctly. You are not supposed to change `default.toml` . You can take it as a template and use it to make your `config.toml`. You then need to drop the file into the root of `boot` partition of the sd card (that's the only partition visible to windows in the whole sd card). A script would then move it to `/etc/pwnagotchi/config.toml` (not visible to windows as its a linux partition and not supported by windows, that's why the whole step is required). After the first boot, you can edit the config over ssh.

LamonJ, 2020-10-13:
awesome, thank you for all your help

thecoder-001, 2020-10-13:
Welcome. You need to save the config as toml. Yaml was the old config format and is now deprecated (and in future its support would be completely dropped). I guess removing the `.txt` extension from end and making the filename `config.toml` should work, but do be on safer side, save your file by using `save as` and then selecting toml from there (if  toml format is not listed there, then just change the extension as you were going to do before).

LamonJ, 2020-10-14:
Ok, man, you have been so helpful and have explained things in a manner that I understand... so thank you.

LamonJ, 2020-10-14:
Bellow is my config.toml file that I put on the boot partion of the sd card. With this config my gothci wont boot up. Any thoughts?

main.name = "PWNserker"
main.lang = "en"
main.confd = "/etc/pwnagotchi/conf.d/"
main.custom_plugins = ""
main.custom_plugin_repos = [
  "https://github.com/evilsocket/pwnagotchi-plugins-contrib/archive/master.zip"
]
main.iface = "mon0"
main.mon_start_cmd = "/usr/bin/monstart"
main.mon_stop_cmd = "/usr/bin/monstop"
main.mon_max_blind_epochs = 50
main.no_restart = false
main.whitelist = [
  "ATT......"
  "ATT........a"
]
main.filter = ""

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "ATT......"
  "ATT........a"
]

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.net-pos.enabled = false
main.plugins.net-pos.api_key = "test"

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.webgpsmap.enabled = false

main.plugins.onlinehashcrack.enabled = false
main.plugins.onlinehashcrack.email = ""
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false
main.plugins.onlinehashcrack.whitelist = []

main.plugins.wpa-sec.enabled = false
main.plugins.wpa-sec.api_key = ""
main.plugins.wpa-sec.api_url = "https://wpa-sec.stanev.org"
main.plugins.wpa-sec.download_results = false
main.plugins.wpa-sec.whitelist = []

main.plugins.wigle.enabled = false
main.plugins.wigle.api_key = ""
main.plugins.wigle.whitelist = []

main.plugins.bt-tether.enabled = true

main.plugins.bt-tether.devices.android-phone.enabled = false
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
main.plugins.bt-tether.devices.android-phone.share_internet = false
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = true
[... 140 lines cut ...]
  "ble.device.connected",
  "ble.device.service.discovered",
  "ble.device.characteristic.discovered",
  "wifi.client.new",
  "wifi.client.lost",
  "wifi.client.probe",
  "wifi.ap.new",
  "wifi.ap.lost",
  "mod.started"
]

fs.memory.enabled = false
fs.memory.mounts.log.enabled = false
fs.memory.mounts.log.mount = "/var/log"
fs.memory.mounts.log.size = "50M"
fs.memory.mounts.log.sync = 60
fs.memory.mounts.log.zram = true
fs.memory.mounts.log.rsync = true

fs.memory.mounts.data.enabled = false
fs.memory.mounts.data.mount = "/var/tmp/pwnagotchi"
fs.memory.mounts.data.size = "10M"
fs.memory.mounts.data.sync = 3600
fs.memory.mounts.data.zram = false
fs.memory.mounts.data.rsync = true

thecoder-001, 2020-10-18:
Sorry for the late response, got very busy in some work. For starters, the first boot takes more time than usual as pwnagotchi needs to generate SSH keys (and as the cpu is not very fast, the process is slow). So just keep it plugged for some time (for some people, its as long as about 30 minutes)  for the first time. 
Also, how are you able to tell that your pwnagotchi is not booting? Is the onboard LED not working too?
I'm going to check the config as soon as I find time (hopefully in the next few hours) and see the problems. Thanks for your patience.

LamonJ, 2020-10-18:
I have it plugged in for an hour before I decided that something is wrong. Also, I renamed my pwnagotchi and the new name never shows up. However, if I just throw the IMG on the SD card and plug it in in about 20 or so minutes the pwn works fine.

LamonJ, 2020-10-20:
were you able to check that config file? Just curious, no rush.

thecoder-001, 2020-10-21:
Finally got some time to check. A few things were formatted incorrectly (maybe just due to pasting in GitHub directly, try to use `code blocks` next time). Also, I spotted a few things missing like the `main.plugins.bt-tether.devices.ios-phone.mac = ""` (if you are going to use bluetooth, this field is required) and commas were missing in the lists. I have corrected those and removed some of the unnecessary bits that are not being changed (you can readd them if needed), as a shorter config is less prone to errors. [Don't worry about the removed parameters, they are still applied with their default values by the pwnagotchi by comparing both files]. Here's the corrected and shortened config:
```toml
main.name = "PWNserker"
main.lang = "en"

main.custom_plugins = ""
main.custom_plugin_repos = [
 "https://github.com/evilsocket/pwnagotchi-plugins-contrib/archive/master.zip"
]

main.whitelist = [
 "ATT......",
 "ATT........a"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
 "ATT......",
 "ATT........a"
]

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.net-pos.enabled = false
main.plugins.net-pos.api_key = "test"

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.webgpsmap.enabled = false

main.plugins.onlinehashcrack.enabled = false
main.plugins.onlinehashcrack.email = ""
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false
main.plugins.onlinehashcrack.whitelist = []

main.plugins.wpa-sec.enabled = false
main.plugins.wpa-sec.api_key = ""
main.plugins.wpa-sec.api_url = "https://wpa-sec.stanev.org"
main.plugins.wpa-sec.download_results = false
main.plugins.wpa-sec.whitelist = []

main.plugins.wigle.enabled = false
main.plugins.wigle.api_key = ""
main.plugins.wigle.whitelist = []

main.plugins.bt-tether.enabled = true

main.plugins.bt-tether.devices.android-phone.enabled = false
main.plugins.bt-tether.devices.android-phone.search_order = 2
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
main.plugins.bt-tether.devices.android-phone.share_internet = false
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = true
main.plugins.bt-tether.devices.ios-phone.search_order = 1
main.plugins.bt-tether.devices.ios-phone.mac = ""
main.plugins.bt-tether.devices.ios-phone.ip = "172.20.10.6"
main.plugins.bt-tether.devices.ios-phone.netmask = 14
main.plugins.bt-tether.devices.ios-phone.interval = 1
main.plugins.bt-tether.devices.ios-phone.scantime = 15
main.plugins.bt-tether.devices.ios-phone.max_tries = 0
main.plugins.bt-tether.devices.ios-phone.share_internet = true
main.plugins.bt-tether.devices.ios-phone.priority = 999

main.plugins.memtemp.enabled = true
main.plugins.memtemp.scale = "celsius"
main.plugins.memtemp.orientation = "horizontal"

main.plugins.paw-gps.enabled = false
main.plugins.paw-gps.ip = ""

main.plugins.gpio_buttons.enabled = false

personality.advertise = true
personality.deauth = false

ui.fps = 0.0
ui.font.name = "DejaVuSansMono" # for japanese: fonts-japanese-gothic
ui.font.size_offset = 0 # will be added to the font size

ui.web.enabled = true
ui.web.address = "0.0.0.0"
ui.web.username = "changeme"
ui.web.password = "changeme"
ui.web.origin = ""
ui.web.port = 8080
ui.web.on_frame = ""

ui.display.enabled = true
ui.display.rotation = 180
ui.display.type = "waveshare_2"
ui.display.color = "black"
```
Try to use this config and report back.
Also attaching a zipped version, just in case.
[config.zip](https://github.com/evilsocket/pwnagotchi/files/5413491/config.zip)

LamonJ, 2020-10-21:
Awesome, thank you for looking at that. I will give it a go and I will let you know. Again thanks

LamonJ, 2020-10-27:
Everything seems to be working. I haven't attempted BT-tethering yet nor have I tried network sharing. I didn't see anywhere on the config for de-auth to be enabled or disabled. Also, I wanted to check my understanding of updating the config. If I drop an updated config right to the SD and plug it back into the pwnagothci it completely reloads everything and rebuilds the keys as well. If I SSH into the gotchi and go to the root, the config.toml isn't there, so would I create a new one or nano into the default.toml? If I do the latter will it reload the keys as well? I apologize I'm just a little confused when on that.

LamonJ, 2020-11-01:
Ok, so when I go to the IP (172.20.10.6:8080) that's in the config for the bt-tethering on my iOS device nothing happens. As a matter of fact, when I tether to my iPhone it seems to stop working altogether. The little time area in the upper right shows all zeros. The BT section of the display shows BT-BE or sometimes C, from what I understand C means connected but again if I go to that IP nothing happens. I am clearly doing something wrong. The Section that shows what mode it is in won't go to AUTO but instead goes to AI, but the time doesn't change in the upper right. It will go to MANU when connected to the computer though. Any input would be much appreciated.

thecoder-001, 2020-11-01:
I'm really sorry for my inability to answer your previous questions in time. School work is kept me pretty busy.
>I didn't see anywhere on the config for de-auth to be enabled or disabled.

It's defined by the parameter `personality.deauth = false` and is on line 84 of the config that I sent earlier.

>If I drop an updated config right to the SD and plug it back into the pwnagothci it completely reloads everything and rebuilds the keys as well. 

Yes, that's reloading everything is the part of the script that moves the config to the right location. The correct way is through SSH.

> If I SSH into the gotchi and go to the root, the config.toml isn't there...

The `config.toml` is present in `/etc/pwnagotchi/config.toml` of Pwnagotchi's filesystem and not directly in root directory.
You can access and edit it by `nano /etc/pwnagotchi/config.toml`.

>...nano into the default.toml
baYou won't want to edit `default.toml` as its overwritten every time when you update your Pwnagotchi. So, it would work but you would lose your config on pwnagotchi's update. So, prefer editing `config.toml`

 >If I do the latter will it reload the keys as well?

No, editing the `config.toml` (or `default.toml`) through SSH wouldn't regenerate the keys.

Now back to current message...
Let's start from the easy part:

>The Section that shows what mode it is in won't go to AUTO but instead goes to AI

Your pwnagotchi is in fact supposed to go to `AI` mode after some amount of time after boot. This is the stage when it really starts to learn. Think of `auto` as a conventional bot which just scans the network, deauthenticates (if you have enabled that) and captures handshakes without learning or improving over time. When it's in `AI` mode, the real fun begins. Its during this time when it learns, gets rewarded and shows its mood.
Now you might be thinking that if AI is the mode it's supposed to be in, then why doesn't it starts from it as soon as boot?
Well that's because of the fact that rpi 0's processor is slow and takes long time (about 20 to 30 minutes) to get all the tensorflow libraries ready for the AI. So not to waste those 20 minutes, Evilsocket made the auto mode to capture handshakes while the AI gets ready in the background. When the AI is fully ready, pwnagotchi switches from auto to ai and starts training.

>...but the time doesn't change in the upper right.

That's because the `ui.fps` parameter is set to 0 in the config. When its zero (which is the default), the display updates only for major changes. The time is simply not considered a major change as it updates every second. When the display updates for a major change (like for indicating new handshake grabbed, change of mood, bluetooth status, etc.), the time counter would update with it.
You can of course change this behavior by modifying the value of `ui.fps` parameter to be higher than 0 in config according to your liking. I wouldn't recommend setting that to too high as it would decrease the lifespan of the ePaper display (its the same reason why its default to 0, as it ensures maximum lifespan).

>The BT section of the display shows BT-BE or sometimes C, from what I understand C means connected

Yes, that means that your Pwnagotchi has successfully paired.

>Ok, so when I go to the IP (172.20.10.6:8080) that's in the config for the bt-tethering on my iOS device nothing happens.

Are you sure that you have enabled hotspot sharing? It looks like that hotspot sharing on your iOS device hasn't been enabled resulting in the issue. [Here's](https://support.apple.com/en-us/HT204023) how to do it on latest iOS version (tip: enable it before powering on your pwnagotchi to ensure that iOS's  hotspot is fully ready to share internet when the Pwnagotchi connects). Try to do that and report back if it works.
Thanks for your patience.

LamonJ, 2020-11-03:
No worries about answering sooner, you have a life like anyone else. I just appreciate the help and don’t mind waiting a few days for an answer. You have answered a LOT and explained it very well. I will get on and play around with that stuff and let you know how it’s going. Again thanks

thejoker3000, 2021-03-14:
@LamonJ Have you been able to fix this issue? 
The easiest way to start is to do this and it's what I did:
1-  Put the config from [this link](https://pwnagotchi.ai/configuration/#initial-configuration) in `/etc/pwnagotchi/config.toml`
2- Make sure that web UI is enabled
3- Plug the device in `data port` (after inserting the SD card into your pwnagotchi)
4- Setup your net adapter using the configurations from [this link](https://pwnagotchi.ai/configuration/#steps-to-complete-on-your-host-the-pc-that-you-are-connecting-the-pi-to)
5- Open your browser and go to `http://10.0.0.2:8080` - Settings - Webgui settings (or something like that, usually at the bottom)
6- Add/Edit the settings you want - click **Save** 
7- Go back to the main page (`http://10.0.0.2:8080`) and click reboot

Wait a few minutes and the device will come back up again. Let me know if that worked.

Travisj16, 2023-02-17:
I'm currently trying to get mine up and running, can you just drag and drop the config.toml into the SD card?

Flipya4real, 2023-12-31:
i am having trouble with the internet sharing power shell script.  It worked at first then i had problems staying connected to my wifi and when i reset my computer my wifi connection is fine but i cant get the sharing to work to the Pwnagotchi 

any suggestions?

JD-2006, 2024-01-12:
@Travisj16 
If you are on Linux then yes. It goes in /etc/pwnagotchi.
If your starting from scratch I would delete original config.toml, then copy default.toml and rename the copy to config.toml.
Make edits for your system.

@Flipya4real 
If you are trying to share your pc internet, it's easier (maybe quicker) to plug into the pi's otg port, and under 'Network Connections' go to to the internet receiving device i.e. Wi-Fi, then Properties, and the tab for sharing, select the pi's USB/RNDIS.
I have had issues if I leave the internet sharing on and rebooting my pc with the Pi still connected.
Best to disable and disconnect Pi before restarting.

therealRadical-Edward, 2024-09-26:
Greetings,
I am having the same problem with editing my config.toml and putting it into the ./Boot directory. I keep getting write permission errors when either editing in nano, emacs, or using touch to create a config.toml file. I have been working on this for hours and am completely stumped. I am trying to turn my display on in the UI section to = true. I guess it was set to false automatically. I have a v4 2.13 waveshare. The pwnagotchi has detected it and I used raspi-config to set the interfaces to SPI. Any help would be greatly appreciated.

Flipya4real, 2024-09-26:
Playing with the config file after initialization is trickier then before
you start up for the first time

I write mine completely and do the initialization boot then modify as
needed with nano or the web ui

I just landed in Napoli, Italy. I am out of country for a week I’ll try to
respond fast like I normally do but I have spotty service

Ciao

John