evilsocket/pwnagotchi #935: [BUG]

fennectech, 2020-10-11:
**Describe the bug**
Pwnagotchi keeps restarting every few minutes   displatying the startup strings
**To Reproduce**
Steps to reproduce the behavior:
1. Turn it on
2. let it boot
3. sometimes it craashes 

**Expected behavior**
It should turn on and pwn

**Environment (please complete the following information):**
 - 1.3.5
 - default shipped with image
 - pi zero w
 - pisugar v1

**Additional context**
Add any other context about the problem here.
There is wierdness in the log.   
[pwnagotchi.log](https://github.com/evilsocket/pwnagotchi/files/5361815/pwnagotchi.log)

fennectech, 2020-10-11:
Additionaly   I did a fsck on the ext4 partition and there were no issues.

LowSkillDeveloper, 2020-10-12:
Have you turn on the watchdog plugin? If so, then turn it off.

fennectech, 2020-10-13:
After clearing out a 5MB pwngrid log file it is working fine.  Sadly i dont have the files any more. So i cant recreate the issue.

debian-user-france1, 2020-11-13:
please close the issue or unname it if you have the same issue again

closed 2021-12-03 by fennectech