evilsocket/pwnagotchi #936: [BUG] 1.3 LCDHAT display issues only after reloading pwnagitchi service

agentages, 2020-10-13:
**Describe the bug**
Unit runs completely fine with 1.3 lcdhat display, when runnung a systemctl restart pwnagotchi by using a GPIO button or doing the same via shell the display will begin to rotate and not display properly (ex text on top, face on bottom.  Flickering, and reverting to an older screen image (uptime 00:05:00 when it should be 00:10:00) 
This continues until a complete system reboot, I have attempted to use the pwnagotchi --clear to resolve but it continues with the display as described.  Connecting to a PC which triggers monitor mode also triggers the display issue.

This will ONLY happen when restarting the service, oe changing the mode.. I can leave it up for hours and never experience the issue until I restart the service. 

**To Reproduce**

1. Start pwnagotchi as normal 
2. Issue a systemcrl restart pwnagotchi or change the mode 
3. Reboot to return to normal 
Watching pwnagotchi --debug shows initializing display (which takes a bit longer than on first boot ~10 seconds more)

A non fatal error us given after initializing the display 
"Dictionary changed size during iteration" I'm unsure if that is relevant. 

**Expected behavior**
Display should continue to function as it does prior to loading.

**Environment:**
 - Pwnagotchi version 1 5 3
 - OS version default image 
 - Type of hardware Pi 0W
 - Any additional hardware: Waveshare 1.3 LCDHAT

**Additional context**
Add any other context about the problem here.

agentages, 2020-10-14:
Attached is a log after changing mode, in this case --manual.  This was after a 3+ hour session with no issues which leads me to believe it's not the display that is causing the issue.

```
2020-10-13 21:54:04,005] [INFO] Added command: /bin/echo -e 'connect DC:DC:E2:5B:CA:91 ' | bluetoothctl to GPIO #20
[2020-10-13 21:54:04,082] [INFO] Added command: sh /home/pi/bltoggle.sh to GPIO #21[2020-10-13 21:54:04,122] [INFO] Added command: sudo pwnagotchi --clear to GPIO #26[2020-10-13 21:54:04,358] [INFO] BT-TETHER: Successfully loaded ...
[2020-10-13 21:54:04,541] [INFO] initializing lcdhat display
[2020-10-13 21:54:04,600] [WARNING] non fatal error while updating view: dictionary changed size during iteration
[2020-10-13 21:54:25,670] [INFO] [ai] loading /root/brain.json
[2020-10-13 21:54:26,532] [INFO] AgentPwn@1a4493edba0a185deaef97d79d4113d81fc98eab16a2486bd5f88b91de7ccfdf (v1.5.3)
[2020-10-13 21:54:26,588] [INFO] entering manual mode ...
[2020-10-13 21:54:27,913] [INFO] web ui available at http://0.0.0.0:8080/
Unhandled exception in thread started by <bound method Server._http_serve of <pwnagotchi.ui.web.server.Server object at 0xb1eebef0>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/web/server.py", line 49, in _http_serve
    app.run(host=self._address, port=self._port, debug=False)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 943, in run
    run_simple(host, port, self, **options)
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 1052, in run_simple
    inner()
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 1005, in inner
    fd=fd,
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 848, in make_server
    host, port, app, request_handler, passthrough_errors, ssl_context, fd=fd
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 740, in __init__
    HTTPServer.__init__(self, server_address, handler)
  File "/usr/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.7/http/server.py", line 137, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
2020-10-13 21:55:03,248] [INFO] the last session lasted 3 hours, 18 minutes, 8 seconds (202 completed epochs, trained for 177), average reward:0.013349171283461997 (min:-0.35 max:1.0087012987012987)
RTNETLINK answers: File exists
[2020-10-13 21:55:08,802] [INFO] Detected a new session and internet connectivity
```

xenDE, 2020-10-15:
> [2020-10-13 21:54:27,913] [INFO] web ui available at http://0.0.0.0:8080/
> Unhandled exception in thread started by <bound method Server._http_serve of <pwnagotchi.ui.web.server.Server object at 0xb1eebef0>>
> ....
> OSError: [Errno 98] Address already in use

looks like, pwnagotchi is already running or not propper closed.

>1.  Start pwnagotchi as normal
> 2. Issue a systemcrl restart pwnagotchi or change the mode
> ...

What is "systemcrl" ??? your own script or a typo?

try "systemctl stop pwnagotchi" and have a look into the process list, if the process was really killed.

if yes: try "systemctl start pwnagotchi"
if no: look in logs why its not killed. your posted log is not complete, it begins on start second pwnagotchi instance. have a look before.