evilsocket/pwnagotchi #937: Can't share connection on MacOS Catalina

Niubbo75, 2020-10-24:
I've create a folder in /home/pi named scripts, there I've put macos_connection_share.sh
I've made it executable running chmod a+x macos_connection_share.sh
Next I've run the script, both from pi user and root but I got the same issue:

`pi@pwnagotchi:~/scipts $ ./macos_connection_share.sh 
ifconfig: option ``-lu' not recognised.
ifconfig: ``--help' gives usage information.
can't find usb interface with ip 10.0.0.1
pi@pwnagochi:~/scipts $ sudo ./macos_connection_share.sh 
ifconfig: option ``-lu' not recognised.
ifconfig: ``--help' gives usage information.
can't find usb interface with ip 10.0.0.1`

What can I do to fix it?

## Steps to Reproduce (for bugs)
1. Create or download macos_connection_share.sh from github
2. Made it executable
3. Run it with ./macos_connection_share.sh
4. Look at the output

## Your Environment

- [ ] Latest pwnagochi image downloaded from official repository

- [ ] Raspberry Pi0 W

- [ ] MacOX 10.15.7

f-eric, 2020-12-13:
The connection sharing script has to be executed on the device from which you want to share the connection, not on the Pwnagotchi/RaspberryPi. So in your situation you'll have to execute it on your Mac.

Niubbo75, 2021-01-09:
Hello, sorry for the delay in my reply, but I have try it only now.
I've run the script on my Mac (now upgraded to Big Sur), I had run it with sudo but I can't share connection anymore, plus, when I connect also my Pi, I can't surf the net (ping google.com or 8.8.8.8 gave only unreacheable destinations both from Pi and Mac).
Should I have to change something in the script?

msirprism, 2021-01-22:
I am struggling with the same issue, any update OP ?

j796160836, 2021-01-24:
Same issue here. I'm figuring why.

Here is excerpt of `ifconfig` on my Macbook.
I have both wifi and ethernet connected on my Macbook.
```
$ ifconfig
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=400<CHANNEL_IO>
	ether xx:xx:xx:xx:xx:xx
	inet6 xxxx::xxx:xxxx:xxxx:xxxx%en0 prefixlen 64 secured scopeid 0x6
	inet6 xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx prefixlen 64 autoconf secured
	inet6 xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx prefixlen 64 autoconf temporary
	inet 192.168.100.201 netmask 0xffffff00 broadcast 192.168.100.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
en7: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=6407<RXCSUM,TXCSUM,VLAN_MTU,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
	ether xx:xx:xx:xx:xx:xx
	inet6 xxxx::xxx:xxxx:xxxx:xxxx%en7 prefixlen 64 secured scopeid 0x15
	inet6 xxxx:xxxx:xxxx:xxxx:xxx:xxxx:xxxx:xxxx prefixlen 64 autoconf secured
	inet6 xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx prefixlen 64 autoconf temporary
	inet 192.168.100.102 netmask 0xffffff00 broadcast 192.168.100.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (1000baseT <full-duplex>)
	status: active
en14: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=6407<RXCSUM,TXCSUM,VLAN_MTU,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
	ether 02:7b:b7:3c:7b:96
	inet6 xxxx::xxxx:xxxx:xxxx:xxxx%en14 prefixlen 64 secured scopeid 0x17
	inet 10.0.0.1 netmask 0xffff0000 broadcast 10.0.255.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (100baseTX <full-duplex>)
	status: active
```

Then here is result of `macos_connection_share.sh`.

```
$sudo ./macos_connection_share.sh
Password:
sharing connecting from upstream interface en0 to usb interface en14 ...
net.inet.ip.forwarding: 1 -> 1
No ALTQ support in kernel
ALTQ related functions disabled
pfctl: pf already enabled
pfctl: Use of -f option, could result in flushing of rules
present in the main ruleset added by the system at startup.
See /etc/pf.conf for further details.

No ALTQ support in kernel
ALTQ related functions disabled
```

Normanras, 2021-04-27:
I don't know how helpful this will be, but 1. this issue seems to go back to the first Pwnagotchi release - lots of reddit posts about getting connection sharing to work on a mac. 

You _could_ look into going into system prefs > sharing and turning on internet sharing to see if that changes anything. (I actually think you need to for bluetooth? That's a separate issue). 

Googling `No ALTQ support in kernel` brings up some StackExchange posts regarding PFSense throwing this issue. One suggestion says to go into System Prefs > Security & Privacy > Firewall and turning the firewall on. I haven't tried this yet, however, so ymmv. 

A slightly separate solution I saw for the `linux_connection_share.sh` was to declare the interfaces you want share. For instance, `sudo ./linux_connection_share.sh [Pwnagotchi Interface Name] [Upstream Interface Name]` Eg. `sudo ./linux_connection_share.sh enp0s20f0u2 wlo1` That suggestion was from [this post.](https://www.reddit.com/r/pwnagotchi/comments/esjlk4/linux_connection_share_woes/). 

If you have any linux machines available, I have had very a very steady working Pwnagotchi when setup, configured, and connected to kali linux. But every time I try to on my mac, I'm met with issues. I would love to see if someone has a succinct solution for this that works for everyone.

dot-iso, 2022-11-16:
I'm also having trouble getting the Pi 0 connected to the internet via MacBook. If I configure Bluetooth I can get it to use my phone data, but I'd like to use my laptop on Wi-Fi so I can install a version of Numpy that allows the AI to work, as well as fix the expired `apt` signing key.

It's also worth noting that I was able to share internet using a Raspberry Pi 3B without issue via an ethernet cable.

Here's what I've tried so far:

### Hardware
* MacBook Pro M1 (macOS 13.0.1 - Ventura)
* Raspberry Pi 0 w with a fresh image
* USB **Data** Cable

### First Attempt
1. System Preferences > Network > RNDIS/Ethernet Gadget > Details
   * TCP/IP
      * Configure IPv4: Manually
      * IP address: `10.0.0.1`
      * Subnet Mask: `255.255.255.0`
      * Router: `10.0.0.1`
1. System Preferences > General > Sharing > Internet Sharing: *Toggle ON*
   * Share your connection from: Wi-Fi
   * To computers using: RNDIS/Ethernet Gadget
1. Plug in completely fresh Pwnagotchi
1. Run `sudo ./macos_connection_sharing.sh en0 en7`
   ```
   sudo ./macos_connection_share.sh en0 en7
   Password:
   sharing connecting from upstream interface en0 to usb interface en7 ...
   net.inet.ip.forwarding: 0 -> 1
   No ALTQ support in kernel
   ALTQ related functions disabled
   pfctl: pf already enabled
   pfctl: Use of -f option, could result in flushing of rules
   present in the main ruleset added by the system at startup.
   See /etc/pf.conf for further details.

   No ALTQ support in kernel
   ALTQ related functions disabled
   ```
1. Once booted
   * My laptop internet stops working
   * I'm unable to SSH into the Pwnagotchi
   * I'm unable to ping Pwnagotchi
   * Pwnagotchi **NOT** connected to the internet
 
### Second Attempt
1. System Preferences > Network > RNDIS/Ethernet Gadget > Details
   * TCP/IP
      * Configure IPv4: Manually
      * IP address: `10.0.0.1`
      * Subnet Mask: `255.255.255.0`
      * Router: `10.0.0.1`
1. System Preferences > General > Sharing > Internet Sharing: *Toggle ON*
   * Share your connection from: Wi-Fi
   * To computers using: RNDIS/Ethernet Gadget
1. Plug in completely fresh Pwnagotchi
1. Once booted
   * My laptop internet stops working
   * I'm unable to SSH into the Pwnagotchi
   * I'm unable to ping Pwnagotchi
   * Pwnagotchi **NOT** connected to the internet

### Third Attempt
1. System Preferences > Network > RNDIS/Ethernet Gadget > Details
   * TCP/IP
      * Configure IPv4: Manually
      * IP address: `10.0.0.1`
      * Subnet Mask: `255.255.255.0`
      * Router: `10.0.0.1`
 1. System Preferences > General > Sharing > Internet Sharing: *Toggle OFF*
 1. Run `sudo ./macos_connection_sharing.sh en0 en7`
   ```
   sudo ./macos_connection_share.sh en0 en7
   Password:
   sharing connecting from upstream interface en0 to usb interface en7 ...
   net.inet.ip.forwarding: 0 -> 1
   No ALTQ support in kernel
   ALTQ related functions disabled
   pfctl: pf already enabled
   pfctl: Use of -f option, could result in flushing of rules
   present in the main ruleset added by the system at startup.
   See /etc/pf.conf for further details.

   No ALTQ support in kernel
   ALTQ related functions disabled
   ```
1. Once booted
   * My laptop internet stops working
   * I'm unable to SSH into the Pwnagotchi
   * I'm unable to ping Pwnagotchi
   * Pwnagotchi **NOT** connected to the internet  

### Fourth Attempt
1. System Preferences > Network > RNDIS/Ethernet Gadget > Details
   * TCP/IP
      * Configure IPv4: Manually
      * IP address: `10.0.0.1`
      * Subnet Mask: `255.255.255.0`
      * Router: `10.0.0.1`
 1. System Preferences > General > Sharing > Internet Sharing: *Toggle OFF*
1. Once booted
   * My laptop internet work
   * I'm able to SSH into the Pwnagotchi
   * I'm able to ping Pwnagotchi
   * Pwnagotchi **NOT** connected to the internet

dot-iso, 2022-11-16:
Ha, of course after I finally write out the post I figure it out.

# Solution

1. Plug in Pwnagotchi
1. System Preferences > Network > RNDIS/Ethernet Gadget > Details
  * TCP/IP
    * Configure IPv4: Manually
    * IP address: `10.0.0.1`
    * Subnet Mask: `255.255.255.0`
    * Router: `10.0.0.1`
 1. System Preferences > General > Sharing > Internet Sharing: *Toggle OFF*
 1. **HERE'S THE TRICK**: System Preferences > Network > Hit the "... V" looking button and choose "Set Service Order". **Make sure Wi-Fi is above RNDIS/Ethernet Gadget**
1. Run `sudo ./macOS_connection_share.sh en0 en7`

rinchen, 2023-05-19:
NB:

- The connections in step 5 in the previous comment should be confirmed by the user by doing an `ifconfig`.  en0 is likely always correct but en7 will be highly user-dependent (it is en11 in my case). To find this, look for the 10.0.0.1 IP address assigned to the interface.
- Enabling the sharing script disabled iCloud Private Relay. To fix this, 
  - `sudo sysctl -w net.inet.ip.forwarding=0` (You could skip this but in the interest of completeness it's here)
  - `sudo pfctl -d` (This disables packet filtering and re-enables Privacy Relay)

michkz, 2023-07-21:
So I did everything you said @rinchen in your post and it seems you'll get the error `No ALTQ support in kernel
ALTQ related functions disabled` regardless. I did notice that `net.inet.ip.forwarding: 1 -> 1` changed from 0 -> 1 and it does work now. I haven't checked if it still works after a reboot though, but thanks for your solution

nethunter, 2025-03-28:
Setting manual IP, and then doing connection sharing in macOS worked perfectly for me.