evilsocket/pwnagotchi #938: [BUG]I run the Linux connection share script,but failed

gili-gili, 2020-10-26:
Cannot find device "enp0s20f0u1"

dohzer, 2020-11-06:
Same for me.

The interface names do not match any of the interfaces reported by my host (including that of the pwnagotchi).

Are we meant to change them to reflect the interface names reported by 'ifconfig'?
If so, it would be nice to have some comments added to the script to explain this.

thecoder-001, 2020-11-07:
Hello!
As @dohzer guessed, you need to manually change the interface names (both the `USB_IFACE` and the `UPSTREAM_IFACE`) values to those matching your `ifconfig` output.
The comments would surely be a nice addition since the current ones only remotely hint to change those values and might cause confusion for new members.

gili-gili, 2020-11-07:
> Hello!
> As @dohzer guessed, you need to manually change the interface names (both the `USB_IFACE` and the `UPSTREAM_IFACE`) values to those matching your `ifconfig` output.
> The comments would surely [...]

Thanks,I’ll try it
:)

dohzer, 2020-11-08:
> Hello!
> As @dohzer guessed, you need to manually change the interface names (both the `USB_IFACE` and the `UPSTREAM_IFACE`) values to those matching your `ifconfig` output.

Yeah, I managed to get it working soon after commenting.
I realised I'd missed the $1/$2 arguments, but I'm still confused as to why there are defaults there? Are they the original author's interface names?
Honestly, reading the Mac version of the script was probably what helped me work out what was going on.

Another reason it was so difficult to get things up and running was that I'd all ready set up an incorrect iptable entry with the tool, and wasn't sure how to remove it, and the script kept complaining about it.

Cheers!

gili-gili, 2020-11-10:
I found something,and it may help you.
iptables -F
service iptables save
service iptables restart
The commands can clear all the rules set by iptables.

dohzer, 2020-11-13:
> I found something,and it may help you.
> iptables -F
> service iptables save
> service iptables restart
> The commands can clear all the rules set by iptables.

Handy to know. Thanks!

sharethewisdom, 2021-01-23:
How do I find the "host interface to use for upstream connection"?

closed 2021-12-17 by gili-gili