evilsocket/pwnagotchi #939: Raspberry Pi 4 Kali Linux + Pwnagotchi

daemonsneverstop, 2020-11-01:
I am very new to Linux but am diving headfirst into learning more about Kali Linux and Raspberry pi, I was able to set up a raspberry pi running kali linux and I was able to successfully create a pwnagotchi. What I do not understand how to do is to set up the host connection sharing. I understand that we have to run the script you listed here but I have no clue how to do that as silly as it may sound. Is it possible for someone to give me a step by step for what exactly I need to type into the command line to get this to work? It would be much appreciated. Thanks.

qkum, 2021-03-21:
Hmm I think you misunderstood something. 

Maybe you are talking about internet sharing over bluetooth? I have not tried that yet myself.

In this pentest space there is not much n00b friendly help. 

Sometimes there are, but then the guides are outdated and you hit a wall.

Only solution is learning to learn to code & master linux.

Then you can figure out how to deal with most problems you self without needing someone to code you a 1 click solution for everything :)

aspman, 2021-03-21:
Correct me if I'm wrong, as I'm just commenting quickly while I have the time, but isn't Kali mentioned as being not supported due to compilation incompatibility anyway? Be better off getting up and running with raspian first I think.

lolzdatguy, 2021-04-14:
You need to change things in /boot/config.txt to allow interfacing over the ethernet port with the computer, keep in mind the ethernet port will not work the way it did before. Follow the pwnagotchi guide for installing pwnagotchi on Linux manually (some of the commands include outdated links for amd64, you need arm64. Find the most current ones). Also DON'T use Kali Linux arm version, it will make it more difficult.