evilsocket/pwnagotchi #941: Warning: New Inky pHAT SSD1608 250x122 not-compatible with built-in driver

Gadgetoid, 2020-11-02:
This isn't a request, but a gentle warning that we've recently launched a revision of the Inky pHAT board that includes a new 250x122 e-Ink display with an SSD1608 driver. However it's still carrying the "Inky pHAT" moniker and we haven't been very proactive about communicating the display differences to customers since:

1. It's an upgrade, nobody is going to complain their display has more pixels than advertised (I hope?)
2. Our own software uses the onboard EEPROM to detect the board type and set it up automagically

As such there may be a bunch of people with an "Inky pHAT" which is *not* compatible with the driver in this project. This can usually be identified by running `python3 -m inky.eeprom` and looking for variants 10, 11, and 12 or a 250x122 resolution.

There's some discussion about a fast update mode for the SSD1608 but it's not something I have the time to investigate at the moment - https://github.com/pimoroni/inky/issues/63

Do reach out if you have any questions!

bdgoodman, 2020-12-27:
@Gadgetoid I am a new customer of the revised Inky pHAT board. It is _hard_ to agree with the rationalization of your first point - _thanks for the extra pixels but communicating change is expected._ To add to the confusion my display is variant 10 and reports the old resolution.

That said, _I do appreciate_ that you posted this warning since it confirmed _I am not crazy_ and pointed in a direction to make it work with pwnagotchi.

I was going to initiate a pull request, but I see the team would rather track using issues. One could imagine making the existing inky definition smarter, but in this case I just defined a new display type and made the other necessary changes to pick it up.

For those that want a quick fix files attached.

[pwnagotchi-display-changes.zip](https://github.com/evilsocket/pwnagotchi/files/5743948/pwnagotchi-display-changes.zip)

Gadgetoid, 2020-12-27:
@bdgoodman Oh I very much agree with you. It isn't hard to contrive some scenario where an unannounced switch of display driver/resolution could really be a problem.

Thanks for taking the initiative and putting together a fix!

kmcdan3944k, 2021-01-09:
@bdgoodman I might be confusing myself with this new display and zip you attached. Installed the new inky drivers via the git-repo and then moved your changed files all into their respected directories without any luck. Service is failing to run

```
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 136, in <module>
    from pwnagotchi.ui.display import Display
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 6, in <module>
    import pwnagotchi.ui.hw as hw
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/__init__.py", line 1, in <module>
    from pwnagotchi.ui.hw.inky import Inky
ImportError: cannot import name 'Inky' from 'pwnagotchi.ui.hw.inky' (/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/inky.py)

```

lildaze, 2021-04-09:
I got a ? I'm kinda new to this stuff and I was wondering how do I copy over the drivers I have the new inkyphat , just don't know where to start can 1 of y'all help me if you can thanks