evilsocket/pwnagotchi #942: [BUG] Bettercap not starting because wlan0/mon0 is available

flow23, 2020-11-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install of v1.5.3 on Pi Zero W
2. Plug in USB cable to USB port
3. Configure interface 10.0.0.2, 255.255.255.0 and 8.8.8.8
4. Connect via ssh
5. tail -f /var/log/pwnagotchi.log

**Expected behavior**
wlan0 interface to be up
bettercap starting mon0

**Screenshots**
pwnagotchi.log
`
[2019-07-10 02:18:36,744] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-07-10 02:18:36,747] [INFO] initializing waveshare v2 display
[2019-07-10 02:18:46,245] [INFO] pwnagotchi@XXX (v1.5.3)
[2019-07-10 02:18:46,260] [INFO] entering auto mode ...
[2019-07-10 02:18:46,299] [INFO] [ai] bootstrapping dependencies ...
[2019-07-10 02:18:46,372] [INFO] waiting for bettercap API to be available ...
[2019-07-10 02:18:46,878] [INFO] web ui available at http://0.0.0.0:8080/
[2019-07-10 02:18:47,464] [INFO] waiting for bettercap API to be available ...
`

`
pi@pwnagotchi:~ $ ifconfig -a
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1341  bytes 87095 (85.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1341  bytes 87095 (85.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::c80c:fff:fe6d:8653  prefixlen 64  scopeid 0x20<link>
        ether ca:0c:0f:6d:86:53  txqueuelen 1000  (Ethernet)
        RX packets 2558  bytes 270709 (264.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1832  bytes 358967 (350.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`

/var/log/messages
`
Jul 10 02:20:03 pwnagotchi systemd[1]: bettercap.service: Service RestartSec=30s expired, scheduling restart.
Jul 10 02:20:03 pwnagotchi systemd[1]: bettercap.service: Scheduled restart job, restart counter is at 5.
Jul 10 02:20:03 pwnagotchi systemd[1]: Stopped bettercap api.rest service..
Jul 10 02:20:03 pwnagotchi systemd[1]: Started bettercap api.rest service..
Jul 10 02:20:04 pwnagotchi bettercap-launcher[1295]: command failed: No such file or directory (-2)
Jul 10 02:20:04 pwnagotchi bettercap-launcher[1295]: WARNING: Terminal colors have been disabled, view will be very limited.
Jul 10 02:20:04 pwnagotchi bettercap-launcher[1295]: bettercap v2.27.1 (built for linux arm with go1.13.1) [type 'help' for a list of commands]
Jul 10 02:20:04 pwnagotchi bettercap-launcher[1295]: no interface matching 'mon0' found.
Jul 10 02:20:04 pwnagotchi systemd[1]: bettercap.service: Main process exited, code=exited, status=1/FAILURE
Jul 10 02:20:04 pwnagotchi systemd[1]: bettercap.service: Failed with result 'exit-code'.
`

dmesg
`
pi@pwnagotchi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.93-Re4son+ (root@kali-pi-dev-0) (gcc version 6.3.0 20170516 (Debian 6.3.0-18)) #1 Wed Jan 8 14:55:22 AEDT 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[... 237 lines cut ...]
[   19.146837] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   19.371664] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   22.113891] Bluetooth: Core ver 2.22
[   22.114074] NET: Registered protocol family 31
[   22.114083] Bluetooth: HCI device and connection manager initialized
[   22.114107] Bluetooth: HCI socket layer initialized
[   22.114121] Bluetooth: L2CAP socket layer initialized
[   22.114171] Bluetooth: SCO socket layer initialized
[   22.146220] Bluetooth: HCI UART driver ver 2.3
[   22.146238] Bluetooth: HCI UART protocol H4 registered
[   22.152540] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   22.152838] Bluetooth: HCI UART protocol Broadcom registered
[   23.258592] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   23.258607] Bluetooth: BNEP filters: protocol multicast
[   23.258632] Bluetooth: BNEP socket layer initialized
`

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.3
 - OS version 
 - Type of hardware Pi Zero W
 - No additional hardware used

**Additional context**
Tried several installs, also changed sd card.

svegeta, 2024-03-03:
Same issue here. Spend the last weeks to try to get it running from scratch, but no luck. Swapping with re4son kernel get the device running, also in monitor mode, but then the AI isn't running anymore, since libavcodec.so.58 can't find the symbol definition for vcsm_init_ex. So, eigther no WiFi or no AI.

tamorak-cfe, 2024-11-05:
Oddly enough, I just stumbled across this bug as well. Did yours start right after doing any OS updates with apt update/upgrade?