evilsocket/pwnagotchi #944: [BUG]

debian-user-france1, 2020-11-08:
After serveral time, the AI don't woks after a reboot:

"Error: the environment passed must have at least the same observation space as the model was trained on."
AssertionError: Error: the environment passed must have at least the same observation space as the model was trained on.
 error message in log

[2020-11-08 13:32:25,025] [ERROR] error while starting AI (Error: the environment passed must have at least the same observation space as the model was trained on.)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/__init__.py", line 48, in load
    "Error: the environment passed must have at least the same observation space as the model was trained on."
AssertionError: Error: the environment passed must have at least the same observation space as the model was trained on.
[2020-11-08 13:32:25,035] [WARNING] [ai] AI not loaded!

I don't understand that...   A reboot didn't change anything, I flashed the image again, now same error.
What is wrong???

I didn't does anything...   I hope, you can help me

debian-user-france1, 2020-11-08:
it "works" after i delete /root/brain.nn

it says, creating new model, after created:

[2020-11-08 14:27:03,559] [INFO] [ai] model created:
INFO:root:      gamma: 0.99
[2020-11-08 14:27:03,560] [INFO]       gamma: 0.99
INFO:root:      n_steps: 1
[2020-11-08 14:27:03,560] [INFO]       n_steps: 1
INFO:root:      vf_coef: 0.25
[2020-11-08 14:27:03,561] [INFO]       vf_coef: 0.25
INFO:root:      ent_coef: 0.01
[2020-11-08 14:27:03,561] [INFO]       ent_coef: 0.01
INFO:root:      max_grad_norm: 0.5
[2020-11-08 14:27:03,562] [INFO]       max_grad_norm: 0.5
INFO:root:      learning_rate: 0.001
[2020-11-08 14:27:03,562] [INFO]       learning_rate: 0.001
INFO:root:      alpha: 0.99
[2020-11-08 14:27:03,563] [INFO]       alpha: 0.99
INFO:root:      epsilon: 1e-05
[2020-11-08 14:27:03,563] [INFO]       epsilon: 1e-05
INFO:root:      verbose: 1
[2020-11-08 14:27:03,564] [INFO]       verbose: 1
INFO:root:      lr_schedule: constant
[2020-11-08 14:27:03,564] [INFO]       lr_schedule: constant
Unhandled exception in thread started by <bound method AsyncTrainer._ai_worker of <pwnagotchi.agent.Agent object at 0x753d2dd0>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 185, in _ai_worker
    obs = self._model.env.reset()
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 52, in reset
    self._save_obs(env_idx, obs)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 71, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (428) into shape (1,503)

debian-user-france1, 2020-11-08:
are important AI files in brain.nn?
and what is this for a bug?

debian-user-france1, 2020-11-08:
after that it is running....
don't know if it learns or not...

debian-user-france1, 2020-11-08:
[2020-11-08 14:33:38,636] [INFO] [ai] learning for 50 epochs ...
ERROR:root:[ai] error while training (could not broadcast input array from shape (428) into shape (1,503))
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 231, in learn
    runner = A2CRunner(self.env, self, n_steps=self.n_steps, gamma=self.gamma)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 310, in __init__
    super(A2CRunner, self).__init__(env=env, model=model, n_steps=n_steps)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/runners.py", line 19, in __init__
    self.obs[:] = env.reset()
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 52, in reset
    self._save_obs(env_idx, obs)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 71, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (428) into shape (1,503)
[2020-11-08 14:33:38,639] [ERROR] [ai] error while training (could not broadcast input array from shape (428) into shape (1,503))
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 231, in learn
    runner = A2CRunner(self.env, self, n_steps=self.n_steps, gamma=self.gamma)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 310, in __init__
    super(A2CRunner, self).__init__(env=env, model=model, n_steps=n_steps)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/runners.py", line 19, in __init__
    self.obs[:] = env.reset()
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 52, in reset
    self._save_obs(env_idx, obs)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 71, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (428) into shape (1,503)

debian-user-france1, 2020-11-13:
sh- of

closed 2020-11-13 by debian-user-france1

XZCendence, 2022-06-10:
> sh- of

did you ever fix this?

arch-user-france1, 2022-06-10:
It never happened again I think. Maybe it got patched, maybe it was just a corrupt brain.nn