evilsocket/pwnagotchi #945: too buggy

debian-user-france1, 2020-11-08:
Pwnagotchi is very buggy (only auto+ai mode)

after some attacks is stop attacking, only searching for networks and bored...
after longer time, it is searching for networks, but nothing is found. Only if you restart pwagotchi, scanning again for networks, attacking again etc.

If it detected a friend, (not every time working, if not working, have to reboot) it says only state of long time... so, it isn't connected to friend...  it is only maybe updating face etc. of friend. You can connect it per Bluetooth or go more often to channel of friend and ask him his state.

there are soo much bugs with the AI and network interface...   Everyone have another bug   
first resolve every bug and fix the auto and ai mode, then add more displays etc. 
I think the AI is learning to nothing...
and is this really legal?
it is kicking a station like my phone, other peoples don't know why there phone havn't internet, in a firma, a NAS crash maybe...
A school is near to me, it got a handshake...   THIS ISN'T LEGAL

debian-user-france1, 2020-11-08:
i'm using it, but YOU HAVE TO RESOLVE THE BUGS AND DO A PASSIVE MODE; ONLY DEAUTHENTICATE THESE NETWORKS...  DEAUTHENTICATE, BUT ONLY LISTEN TO STATION AND GIVE THE STATION CONNECTION TO THE ROUTER etc.

PLEASE FIX THAT

xenDE, 2020-11-08:
"after some attacks is stop attacking,...."
"THIS ISN'T LEGAL"

Please stop this, if it's not legal in your country and you know it.

"PLEASE FIX THAT"
this is a private open source project. you can fix it if something is not working.

if you find a bug and want to report, then report in the right format.

please delete this issue with the reporting of your known illegal activities.

debian-user-france1, 2020-11-08:
I tried to fix the 100000000000000 scripts, but.....
don't understand comments or it doesn't have..
and yes, i stoped...

closed 2020-11-08 by debian-user-france1