evilsocket/pwnagotchi #946: [BUG]   Wrong installation guide in pwnagotchi.ai for linux (kali linux)

debian-user-france1, 2020-11-08:
I tried to install pwnagotchi on my kali linux os, but can't install requirements

wget "https://github.com/evilsocket/pwnagotchi/archive/v1.4.3.zip"
unzip v1.4.3.zip
cd pwnagotchi-1.4.3
# this will install the requirements and pwnagotchi itself
**sudo pip3 install -r requirements.txt
sudo pip3 install .**

Collecting crypto==1.4.1
  Using cached crypto-1.4.1-py2.py3-none-any.whl (18 kB)
Collecting requests==2.21.0
  Using cached requests-2.21.0-py2.py3-none-any.whl (57 kB)
Collecting PyYAML==5.1
  Using cached PyYAML-5.1.tar.gz (274 kB)
Collecting scapy==2.4.3
  Using cached scapy-2.4.3.tar.gz (905 kB)
Collecting gym==0.14.0
  Using cached gym-0.14.0.tar.gz (1.6 MB)
Collecting scipy==1.3.1
  Using cached scipy-1.3.1.tar.gz (23.6 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /tmp/tmp1ydy_max prepare_metadata_for_build_wheel /tmp/tmp0a3tgwcu                                                             
         cwd: /tmp/pip-install-3s7_7m21/scipy                                                                                                                 
    Complete output (156 lines):                                                                                                                              
    lapack_opt_info:                                                                                                                                          
    lapack_mkl_info:                                                                                                                                          
    customize UnixCCompiler                                                                                                                                   
      libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib', '/usr/lib/x86_64-linux-gnu']                                                 
      NOT AVAILABLE                                                                                                                                           
                                                                                                                                                              
    openblas_lapack_info:                                                                                                                                     
    customize UnixCCompiler                                                                                                                                   
    customize UnixCCompiler                                                                                                                                   
      libraries openblas not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib', '/usr/lib/x86_64-linux-gnu']                                               
      NOT AVAILABLE                                                                                                                                           
                                                                                                                                                              
    openblas_clapack_info:                                                                                                                                    
    customize UnixCCompiler                                                                                                                                   
    customize UnixCCompiler                                                                                                                                   
      libraries openblas,lapack not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib', '/usr/lib/x86_64-linux-gnu']                                        
      NOT AVAILABLE                                                                                                                                           
                                                                                                                                                              
    atlas_3_10_threads_info:                                                                                                                                  
    Setting PTATLAS=ATLAS                                                                                                                                     
    customize UnixCCompiler                                                                                                                                   
      libraries tatlas,tatlas not found in /usr/local/lib                                                                                                     
    customize UnixCCompiler                                                                                                                                   
      libraries lapack_atlas not found in /usr/local/lib                                                                                                      
    customize UnixCCompiler                                                                                                                                   
      libraries tatlas,tatlas not found in /usr/lib64                                                                                                         
    customize UnixCCompiler                                                                                                                                   
      libraries lapack_atlas not found in /usr/lib64                                                                                                          
    customize UnixCCompiler                                                                                                                                   
      libraries tatlas,tatlas not found in /usr/lib                                                                                                           
    customize UnixCCompiler                                                                                                                                   
      libraries lapack_atlas not found in /usr/lib                                                                                                            
    customize UnixCCompiler                                                                                                                                   
      libraries tatlas,tatlas not found in /usr/lib/x86_64-linux-gnu                                                                                          
    customize UnixCCompiler                                                                                                                                   
      libraries lapack_atlas not found in /usr/lib/x86_64-linux-gnu                                                                                           
    <class 'numpy.distutils.system_info.atlas_3_10_threads_info'>                                                                                             
      NOT AVAILABLE                                                                                                                                           
                                                                                                                                                              
    atlas_3_10_info:                                                                                                                                          
    customize UnixCCompiler                                                                                                                                   
      libraries satlas,satlas not found in /usr/local/lib                                                                                                     
[... 93 lines cut ...]
      File "/tmp/tmp1ydy_max", line 280, in <module>                                                                                                          
        main()                                                                                                                                                
[... cut ...]
    numpy.distutils.system_info.NotFoundError: No lapack/blas resources found.                                                                                
    ----------------------------------------                                                                                                                  
ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmp1ydy_max prepare_metadata_for_build_wheel /tmp/tmp0a3tgwcu Check the logs for full command output.             

please update installation guide and help me to install this.-

LuckyNumberZero, 2020-11-17:
It looks like you're missing other requirements and that the first pip install command didn't complete right.
I had to install a couple things with apt to get 1.5.3 to build.

debian-user-france1, 2020-11-17:
Can you show me these?
I'm using Ubuntu. How did you found requirements? Googled?

debian-user-france1, 2020-11-17:
Some much libarys are missing...

rcgoodstuf, 2020-12-06:
> It looks like you're missing other requirements and that the first pip install command didn't complete right.
> I had to install a couple things with apt to get 1.5.3 to build.

ee0fa1f50c215863b25a61c62d37067b9b9a290bIP: 10.0.0.1

Netmask: 255.255.255.0

Gateway: 10.0.0.1

DNS (if required): 8.8.8.8 (or whatever)

If everything’s been configured properly, you will now be able to ping either 10.0.0.2 or pwnagotchi.local
 Pwnagotchi, pwnagotchi.local won’t work. Instead, try your unit’s host name coff get32.000.000.02..

 studious-octo-engine

rcgoodstuf, 2020-12-06:
Ssh

debian-user-france1, 2020-12-08:
?!

funkymonkey5150, 2020-12-08:
Having allot of issues I can’t seem to make it work I need a new clean
image and reinstall everything is everything list

gumanov, 2020-12-22:
Having the same issue.
Anyone know what else needs to be installed for the build to work?

Trying to install pwnagotchi v1.5.3 on kali, pi4.

gumanov, 2020-12-29:
Soooo for anyone who is interested, here is how I got pwnagotchi v1.5.3 up and running on Kali Linux 2020.4 (aarch64), Raspberry Pi4:
Kali Image: kali-linux-2020.4-rpi4-nexmon-64.img.xz

This version of Kali Linux runs Python v3.8.x by default. Pwnagotchi is meant to be built on Python 3.7.x.
I managed to get it installed on both Python 3.8.x and Python 3.7.x.
When run on Py3.8.x the AI does not load. The error is:
[ERROR] error while starting AI (No module named 'tensorflow.contrib')
This is because "tensorflow.contrib" was removed in tensorflow 2.x.x.

So I would recommend avoiding Python 3.8.x for now, and building pwnagotchi on Python 3.7.x.
The below instructions are for Python 3.7.x. I can share the 3.8.x instructions for anyone interested, but again the AI will not work on 3.8.x.

_____________________________________
Flash the Kali image onto the microSD card.

Add the following at the bottom of config.txt on the boot partition:
#pwnagotchi
dtoverlay=dwc2
dtoverlay=spi1-3cs
dtparam=spi=on
dtparam=i2c_arm=on
dtparam=i2c1=on
gpu_mem=16
_____________________________________

Place microSD into the Pi3/4 and Boot the OS.
Run the Pi on a LAN with internet access.
SSH in, or hook up an HDMI display and keyboard/mouse to get access to the terminal.
_____________________________________

Install pre-compiled aarch64 BetterCap binary:

wget "https://github.com/bettercap/bettercap/releases/download/v2.28/bettercap_linux_aarch64_v2.28.zip"
unzip bettercap_linux_aarch64_v2.28.zip
sudo mv bettercap /usr/bin/
sudo bettercap -eval "caplets.update; ui.update; quit"
_____________________________________

Install pre-compiled aarch64 PwnGrid binary:

wget "https://github.com/evilsocket/pwngrid/releases/download/v1.10.3/pwngrid_linux_aarch64_v1.10.3.zip"
unzip pwngrid_linux_aarch64_v1.10.3.zip
sudo mv pwngrid /usr/bin/
sudo pwngrid -generate -keys /etc/pwnagotchi
_____________________________________

Install Pwnagotchi dependencies

sudo apt-get update
sudo apt-get -y install gfortran
sudo apt-get -y install libopenmpi-dev
sudo apt-get -y install libdbus-1-dev libdbus-glib-1-dev
sudo apt-get -y install liblapack-dev libhdf5-dev libc-ares-dev libeigen3-dev
sudo apt-get -y install libatlas-base-dev libopenblas-dev libblas-dev

-----------

Install Python 3.7.x
Python 3.7.x Dependencies
sudo apt-get install -y wget build-essential checkinstall
sudo apt-get install -y libreadline-gplv2-dev libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev libffi-dev zlib1g-dev

Add the repo for Python3.7
sudo nano /etc/apt/sources.list
ADD to bottom:
----
deb http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal main
deb-src http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal main
----

[... 161 lines cut ...]
KNOWN ISSUE:
Driver for the built-in wifi card (brcmfmac) keeps crashing
This is a known issue with pwnagotchi.
The pwnagotchi scripts typically reboot the pi if the driver fails and won't reload.
https://github.com/evilsocket/pwnagotchi/issues/267

Possible Solutions:
https://github.com/evilsocket/pwnagotchi/issues/267#issuecomment-571025443
https://github.com/raspberrypi/linux/issues/2453#issuecomment-645963598

I personally run the Pi4 with an Alfa USB WiFi Module attached.
This uses a completely separate driver which does not have crashing issues.

To use an external wifi card with the RPi3/4:
Modify: /boot/config.txt
Add the following line at the bottom of the file:
dtoverlay=pi3-disable-wifi

This will disable the onboard internal wifi, and the external USB wifi will become device wlan0.

There are also ways to keep both enabled, where the external would be wlan1.
You'd have to modify the bettercap/pwnagotchi launcher scripts, the /usr/bin/pwnlib file, and the pwnagotchi-auto/manual caplets to use wlan1/mon1 instead.

The better alternative at that point is to maybe find a way to set the onboard wifi to wlan1, and set the usb wifi device to wlan0 (?).
That way you wouldn't have to modify any pwnagotchi/bettercap files.

debian-user-france1, 2020-12-30:
OMG, thanks

debian-user-france1, 2020-12-30:
I'll try it a bit later

closed 2020-12-31 by debian-user-france1

XZCendence, 2022-06-10:
> Soooo for anyone who is interested, here is how I got pwnagotchi v1.5.3 up and running on Kali Linux 2020.4 (aarch64), Raspberry Pi4:
> Kali Image: kali-linux-2020.4-rpi4-nexmon-64.img.xz
> 
> This version of Kali Linux runs Python v3.8.x by default. Pwnagotchi is meant to be built on Python 3.7.x.
> I managed to get it installed on both Python 3.8.x and Python 3.7.x.
> When run on Py3.8.x the AI does not load. The error is:
> [ERROR] error while starting AI (No module named 'tensorflow.contrib')
> This is because "tensorflow.contrib" was removed in tensorflow 2.x.x (The minimum tensorflow version with Py3.8 support is 2.2.x).
> 
> So I would recommend avoiding Python 3.8.x for now, and building pwnagotchi on Python 3.7.x.
> The below instructions are for Python 3.7.x. I can share the 3.8.x instructions for anyone interested, but again the AI will not work on 3.8.x.
> 
> _____________________________________
> Flash the Kali image onto the microSD card.
> 
> Add the following at the bottom of config.txt on the boot partition:
> #pwnagotchi
> dtoverlay=dwc2
> dtoverlay=spi1-3cs
> dtparam=spi=on
> dtparam=i2c_arm=on
> dtparam=i2c1=on
> gpu_mem=16
> _____________________________________
> 
> Place microSD into the Pi3/4 and Boot the OS.
> Run the Pi on a LAN with internet access.
> SSH in, or hook up an HDMI display and keyboard/mouse to get access to the terminal.
> _____________________________________
> 
> Install pre-compiled aarch64 BetterCap binary:
> 
> wget "https://github.com/bettercap/bettercap/releases/download/v2.28/bettercap_linux_aarch64_v2.28.zip"
> unzip bettercap_linux_aarch64_v2.28.zip
> sudo mv bettercap /usr/bin/
> sudo bettercap -eval "caplets.update; ui.update; quit"
> _____________________________________
> 
> Install pre-compiled aarch64 PwnGrid binary:
> 
> wget "https://github.com/evilsocket/pwngrid/releases/download/v1.10.3/pwngrid_linux_aarch64_v1.10.3.zip"
> unzip pwngrid_linux_aarch64_v1.10.3.zip
> sudo mv pwngrid /usr/bin/
> sudo pwngrid -generate -keys /etc/pwnagotchi
> _____________________________________
> 
> Install Pwnagotchi dependencies
> 
> sudo apt-get update
> sudo apt-get -y install gfortran
> sudo apt-get -y install libopenmpi-dev
> sudo apt-get -y install libdbus-1-dev libdbus-glib-1-dev
> sudo apt-get -y install liblapack-dev libhdf5-dev libc-ares-dev libeigen3-dev
> sudo apt-get -y install libatlas-base-dev libopenblas-dev libblas-dev
> 
> -----------
> 
> Install Python 3.7.x
> Python 3.7.x Dependencies
> sudo apt-get install -y wget build-essential checkinstall
> sudo apt-get install -y libreadline-gplv2-dev libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev libffi-dev zlib1g-dev
> 
> Add the repo for Python3.7: 
> sudo nano /etc/apt/sources.list
> ADD to bottom:
> ```
> deb http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal main
> deb-src http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal main
> ```
> 
[... 174 lines cut ...]
> https://github.com/evilsocket/pwnagotchi/issues/267#issuecomment-571025443
> https://github.com/raspberrypi/linux/issues/2453#issuecomment-645963598
> 
> I personally run the Pi4 with an Alfa USB WiFi Module attached.
> This uses a completely separate driver which does not have crashing issues.
> 
> To use an external wifi card with the RPi3/4:
> Modify: /boot/config.txt
> Add the following line at the bottom of the file:
> ```
> dtoverlay=pi3-disable-wifi
> ```
> This will disable the onboard internal wifi, and the external USB wifi will become device wlan0.
> 
> There are also ways to keep both enabled, where the external would be wlan1.
> You'd have to modify the bettercap/pwnagotchi launcher scripts, the /usr/bin/pwnlib file, and the pwnagotchi-auto/manual caplets to use wlan1/mon1 instead.
> 
> The better alternative at that point is to maybe find a way to set the onboard wifi to wlan1, and set the usb wifi device to wlan0 (?).
> That way you wouldn't have to modify any pwnagotchi/bettercap files.
> 
> 
> 
> 

I tried this out but I couldn't get AI to start properly, it keeps throwing that one error "could not broadcast input array" back when 5ghz channels were not supported

XZCendence, 2022-06-11:
> Soooo for anyone who is interested, here is how I got pwnagotchi v1.5.3 up and running on Kali Linux 2020.4 (aarch64), Raspberry Pi4: Kali Image: kali-linux-2020.4-rpi4-nexmon-64.img.xz
> 
> This version of Kali Linux runs Python v3.8.x by default. Pwnagotchi is meant to be built on Python 3.7.x. I managed to get it installed on both Python 3.8.x and Python 3.7.x. When run on Py3.8.x the AI does not load. The error is: [ERROR] error while starting AI (No module named 'tensorflow.contrib') This is because "tensorflow.contrib" was removed in tensorflow 2.x.x (The minimum tensorflow version with Py3.8 support is 2.2.x).
> 
> So I would recommend avoiding Python 3.8.x for now, and building pwnagotchi on Python 3.7.x. The below instructions are for Python 3.7.x. I can share the 3.8.x instructions for anyone interested, but again the AI will not work on 3.8.x.
> 
> Flash the Kali image onto the microSD card.
> 
> Add the following at the bottom of config.txt on the boot partition: #pwnagotchi dtoverlay=dwc2 dtoverlay=spi1-3cs dtparam=spi=on dtparam=i2c_arm=on dtparam=i2c1=on gpu_mem=16
> 
> Place microSD into the Pi3/4 and Boot the OS. Run the Pi on a LAN with internet access. SSH in, or hook up an HDMI display and keyboard/mouse to get access to the terminal.
> 
> Install pre-compiled aarch64 BetterCap binary:
> 
> wget "https://github.com/bettercap/bettercap/releases/download/v2.28/bettercap_linux_aarch64_v2.28.zip" unzip bettercap_linux_aarch64_v2.28.zip sudo mv bettercap /usr/bin/ sudo bettercap -eval "caplets.update; ui.update; quit"
> 
> Install pre-compiled aarch64 PwnGrid binary:
> 
> wget "https://github.com/evilsocket/pwngrid/releases/download/v1.10.3/pwngrid_linux_aarch64_v1.10.3.zip" unzip pwngrid_linux_aarch64_v1.10.3.zip sudo mv pwngrid /usr/bin/ sudo pwngrid -generate -keys /etc/pwnagotchi
> 
> Install Pwnagotchi dependencies
> 
> sudo apt-get update sudo apt-get -y install gfortran sudo apt-get -y install libopenmpi-dev sudo apt-get -y install libdbus-1-dev libdbus-glib-1-dev sudo apt-get -y install liblapack-dev libhdf5-dev libc-ares-dev libeigen3-dev sudo apt-get -y install libatlas-base-dev libopenblas-dev libblas-dev
> 
> Install Python 3.7.x Python 3.7.x Dependencies sudo apt-get install -y wget build-essential checkinstall sudo apt-get install -y libreadline-gplv2-dev libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev libffi-dev zlib1g-dev
> 
> Add the repo for Python3.7: sudo nano /etc/apt/sources.list ADD to bottom:
> 
> ```
> deb http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal main
> deb-src http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal main
> ```
> 
> sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys BA6932366A755776 sudo apt-get update sudo apt-get install -y python3.7 python3.7-dev
> 
> Update some Python3.7 modules sudo python3.7 -m pip install --upgrade pip sudo python3.7 -m pip install --upgrade wheel sudo python3.7 -m pip install --upgrade setuptools
> 
> Install TensorFlow 1.15.0: Pwnagotchi requires tensorflow v1.13.1, but I was unable to find any python whl's for that version (aarch64, python 3.7). Version 1.15.0 seems to do just fine. I did eventually find instructions for compiling v1.13.1, but never tested it: https://gist.github.com/apivovarov/6274e4f0afd294c112102868c9e84d94 So I would suggest to stick to 1.15.0 unless you wanna give it a shot yourself. (Note: I've read that it takes roughly ~1.5 hours to compile tensorflow on a Pi3 from scratch). Additional v1.15.0 info: https://github.com/PINTO0309/Tensorflow-bin
> 
> Install gdown to download the whl file from Google Drive: python3.7 -m pip install gdown sudo cp ~/.local/bin/gdown /usr/local/bin/gdown
> 
> Download and install tensorflow: gdown https://drive.google.com/uc?id=1D9IT2WC2mrWOixsNKohxiJJIlY2IzmQQ sudo python3.7 -m pip install tensorflow-1.15.0-cp37-cp37m-linux_aarch64.whl
> 
> Due to a recent change to gfortran version 10, building scipy v1.3.1 will fail when using any 10.x version. Let's setup to use gfortran-9 by default instead sudo apt-get install gcc-9 gfortran-9 sudo unlink /usr/bin/gcc sudo ln -s /usr/bin/gcc-9 /usr/bin/gcc sudo unlink /usr/bin/gfortran sudo ln -s /usr/bin/gfortran-9 /usr/bin/gfortran
> 
> Compile Pwnagotchi from latest source code:
> 
> wget "https://github.com/evilsocket/pwnagotchi/archive/v1.5.3.zip" unzip v1.5.3.zip cd pwnagotchi-1.5.3
> 
> sudo nano requirements.txt Set tensorflow version to 1.15.0 and tensorflow-estimator version to 1.15.1 If you have issues building scipy v1.3.1, Remove version hard-coding for: scipy, numpy This will install the latest versions of those modules. After swapping to gfrotran v9.x, I had no issues with scipy v1.3.1.
> 
> sudo python3.7 -m pip install -r requirements.txt sudo python3.7 -m pip install .
> 
> Congratulations, pwngotchi is installed. Follow further instructions below for various fixes:
> 
> Enable services for start-on-boot sudo systemctl enable bettercap pwngrid-peer pwnagotchi
> 
> Fix permissions errors sudo chmod 755 /usr/bin/bettercap sudo chown root:root /usr/bin/bettercap sudo chmod 755 /usr/bin/bettercap-launcher sudo chmod 755 /usr/bin/pwngrid sudo chown root:root /usr/bin/pwngrid sudo chmod 755 /usr/local/bin/pwnagotchi sudo chown root:root /usr/local/bin/pwnagotchi sudo chmod 711 /usr/bin/pwnagotchi-launcher
> 
[... cut ...]
> There are also ways to keep both enabled, where the external would be wlan1. You'd have to modify the bettercap/pwnagotchi launcher scripts, the /usr/bin/pwnlib file, and the pwnagotchi-auto/manual caplets to use wlan1/mon1 instead.
> 
> The better alternative at that point is to maybe find a way to set the onboard wifi to wlan1, and set the usb wifi device to wlan0 (?). That way you wouldn't have to modify any pwnagotchi/bettercap files.

Could you help me install this on kali 2022.2? a lot of this doesn't work anymore
Thanks

curious-ninja, 2022-06-11:
Can you be more specific with what exactly doesn't work throughout the entire setup process?

At which step do you run into issues?

arch-user-france1, 2022-06-12:
pip install -r requirements.txt didn't work

yassinebkr, 2024-03-10:
> Soooo for anyone who is interested, here is how I got pwnagotchi v1.5.3 up and running on Kali Linux 2020.4 (aarch64), Raspberry Pi4: Kali Image: kali-linux-2020.4-rpi4-nexmon-64.img.xz
>
> This [...]

you're a goat