evilsocket/pwnagotchi #954: Config.toml

Busdesto, 2020-11-22:
Hi my pwnagotchi works perfectly fine with the waveshare 2 black and white display (the one with the red tab on it) when using defaults.toml however no matter how I decide to input my config.toml (either through dropping the file in windows or ssh in Linux and changing the directory) the waveshare screen does not work with the config.toml and I think the entire pwnagotchi is not working with the config.toml as well. I have checked to see if the formatting of the config.toml is wrong but it isn't and wont work with it. It would be nice to have my home networks whitelisted so I don't keep pwning them everytime I leave my pwnagotchi on. Any help would be greatly appreciated thanks!

thecoder-001, 2020-11-22:
Hello!
Can you share your config? I would verify it as soon as I get some time. I guess something is wrong there, stopping the pwnagotchi from starting.
Cheers

Busdesto, 2020-11-23:
`# WARNING WARNING WARNING WARNING
#
# This file is recreated with default settings on every pwnagotchi restart,
# use /etc/pwnagotchi/config.yml to configure this unit.
#
#
# main algorithm configuration
main:
    # currently implemented: en (default), de, el, fr, it, mk, nl, ru, se, pt-BR, es, pt
    lang: en
    # custom plugins path, if null only default plugins with be loaded
    custom_plugins:
    # which plugins to load and enable
    plugins:
        grid:
            enabled: true
            report: false # don't report pwned networks by default!
            exclude: # do not report the following networks (accepts both ESSIDs and BSSIDs)
                - 'CawdonHouse'
                - 'CawdonHouseGuest'
                - 'NETGEAR72'
                - 'NEATGEAR72-5G'
        auto-update:
            enabled: false
            interval: 12 # every 12 hours
            install: true # if false, it will only warn that updates are available, if true it will install them
        auto-backup:
            enabled: false
            interval: 1 # every day
            files:
                - /root/brain.nn
                - /root/brain.json
                - /root/.api-report.json
                - /root/handshakes/
                - /etc/pwnagotchi/
                - /var/log/pwnagotchi.log
            commands:
                - 'tar czf /root/pwnagotchi-backup.tar.gz {files}'
        net-pos:
            enabled: false
            api_key: 'test'
        gps:
            enabled: false
            speed: 19200
            device: /dev/ttyUSB0
        twitter:
            enabled: false
            consumer_key: aaa
            consumer_secret: aaa
            access_token_key: aaa
            access_token_secret: aaa
        onlinehashcrack:
            enabled: false
            email: ~
        wpa-sec:
            enabled: false
            api_key: ~
            api_url: "https://wpa-sec.stanev.org"
        wigle:
            enabled: false
            api_key: ~
        screen_refresh:
            enabled: false
            refresh_interval: 50
        quickdic:
            enabled: false
            wordlist_folder: /opt/wordlists/
        AircrackOnly:
            enabled: false
        bt-tether:
[... 138 lines cut ...]
    # api scheme://hostname:port username and password
    scheme: http
    hostname: localhost
    port: 8081
    username: pwnagotchi
    password: pwnagotchi
    # folder where bettercap stores the WPA handshakes, given that
    # wifi.handshakes.aggregate will be set to false and individual
    # pcap files will be created in order to minimize the chances
    # of a single pcap file to get corrupted
    handshakes: /handshakes
    # events to mute in bettercap's events stream
    silence:
        - ble.device.new
        - ble.device.lost
        - ble.device.disconnected
        - ble.device.connected
        - ble.device.service.discovered
        - ble.device.characteristic.discovered
        - wifi.client.new
        - wifi.client.lost
        - wifi.client.probe
        - wifi.ap.new
        - wifi.ap.lost
        - mod.started`

thecoder-001, 2020-11-26:
hmm, you seem to be using the old yml format in your config file instead of toml. Here's the corrected config. Also from next time, please try to use `code blocks`.
```toml
main.name = "Pwnagotchi"
main.lang = "en"
main.confd = "/etc/pwnagotchi/conf.d/"
main.custom_plugins = ""
main.custom_plugin_repos = [
  "https://github.com/evilsocket/pwnagotchi-plugins-contrib/archive/master.zip"
]
main.iface = "mon0"
main.mon_start_cmd = "/usr/bin/monstart"
main.mon_stop_cmd = "/usr/bin/monstop"
main.mon_max_blind_epochs = 50
main.no_restart = false
main.whitelist = [
  "CawdonHouse",
  "CawdonHouseGuest",
  "NETGEAR72",
  "NEATGEAR72-5G"
]
main.filter = ""

main.plugins.grid.enabled = true
main.plugins.grid.report = false
main.plugins.grid.exclude = [
  "CawdonHouse",
  "CawdonHouseGuest",
  "NETGEAR72",
  "NEATGEAR72-5G"
]

main.plugins.auto-update.enabled = false
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.net-pos.enabled = false
main.plugins.net-pos.api_key = "test"

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.webgpsmap.enabled = false

main.plugins.onlinehashcrack.enabled = false
main.plugins.onlinehashcrack.email = ""
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false
main.plugins.onlinehashcrack.whitelist = []

main.plugins.wpa-sec.enabled = false
main.plugins.wpa-sec.api_key = ""
main.plugins.wpa-sec.api_url = "https://wpa-sec.stanev.org"
main.plugins.wpa-sec.download_results = false
main.plugins.wpa-sec.whitelist = []

main.plugins.wigle.enabled = false
main.plugins.wigle.api_key = ""
main.plugins.wigle.whitelist = []

main.plugins.bt-tether.enabled = false

main.plugins.bt-tether.devices.android-phone.enabled = false
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
[... 149 lines cut ...]
  "wifi.client.probe",
  "wifi.ap.new",
  "wifi.ap.lost",
  "mod.started"
]

fs.memory.enabled = false
fs.memory.mounts.log.enabled = false
fs.memory.mounts.log.mount = "/var/log"
fs.memory.mounts.log.size = "50M"
fs.memory.mounts.log.sync = 60
fs.memory.mounts.log.zram = true
fs.memory.mounts.log.rsync = true

fs.memory.mounts.data.enabled = false
fs.memory.mounts.data.mount = "/var/tmp/pwnagotchi"
fs.memory.mounts.data.size = "10M"
fs.memory.mounts.data.sync = 3600
fs.memory.mounts.data.zram = false
fs.memory.mounts.data.rsync = true
```
Attaching a compressed version, just in case.
[config.toml.zip](https://github.com/evilsocket/pwnagotchi/files/5600463/config.toml.zip)
Try this one and report back.
Cheers

Busdesto, 2020-11-29:
Thanks very much everything seems to be working as expected