evilsocket/pwnagotchi #958: Use GPS as clock source

sturles, 2021-01-28:
My pwnagotchi's clock is far off due to frequent reboots and some downtime now and then.  I could connect a RTC module, but why not get exact time from my GPS module instead?

NTP can read the time from gpsd if it is running, but I understand that pwnagotchi read GPS data directly from the serial output.  This will probably not work well with another process wanting to read and write the same TTY.

How could this be solved?  Can pwnagotchi read the time directly from the GPS output and set it, or could it read position data from gpsd instead of directly from the GPS serial device?

virtadpt, 2021-01-28:
It's pretty easy to do, I do it on my Pwnagotchi for that reason.  Here's how I do it:

https://drwho.virtadpt.net/archive/2018-11-28/build-your-own-time-server-with-a-gps-receiver/

sturles, 2021-02-03:
How do you get gpsd and bettercap to cooperate?  As soon as my pwnagotchi enters auto mode, gpsd no longer receive any data, and no location data is shown on my pwnagotchi either.  Both processes are fighting over the data coming in via /dev/ttyAMA0:
```
$ sudo lsof /dev/ttyAMA0 
COMMAND     PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
gpsd        424 gpsd    7u   CHR 204,64      0t0 6018 /dev/ttyAMA0
bettercap 24762 root   10u   CHR 204,64      0t0 6018 /dev/ttyAMA0
```

It would be better if bettercap could interface with gpsd instead of reading raw GPS data from the serial device.

virtadpt, 2021-02-03:
Per #896 Bettercap never successfully accessed my GPS, so that has not been an issue.  At least for me, my decision to use gpsd for clock synch was driven by Bettercap's GPS support not working regardless of what I did to troubleshoot it, so I decided to use it for something useful.

I have no idea why Bettercap doesn't just connect to gpsd like other software does.  It's a misfeature in quite a few posteriors.

virtadpt, 2021-02-03:
In theory, this bug in Bettercap is fixed:

https://github.com/bettercap/bettercap/issues/580

https://github.com/bettercap/bettercap/issues/805

In practice, it'll require a new release of the Pwnagotchi firmware image, or surgery on an existing one.

davenicoll, 2021-03-29:
I use GPS on all of my pwnagotchi's (they travel a fair bit), so I'll see if I can get this working.