evilsocket/pwnagotchi #959: BT-Tether Stuck at 'NF'

Declan0811, 2021-02-03:
Phone - iPhone
Most recent version (I forget which actual version number it is, all I know is that it is the most recent)

So to pair my phone I did all of the necessary steps:

boot the RPi0W using the DATA PORT
Connected to the RPi0W with putty
Paired the iPhone with the RPi0W with these steps:
- sudo bluetoothctl
- scan on
- put the iPhone in discovery mode (settings - bluetooth)
- wait until the RPi0W found the iPhone then
- pair 40:9C:28:XX:XX:XX (BT mac address of the iPhone)
- confirmed the pair process
- trust 40:9C:28:XX:XX:XX (BT mac address of the iPhone)
- stop the scanning process scan off
- exit form bluetoothctl exit
- now in the BT paired device list of the iPhone I can see the RPi0W (the hostname)
- start the pwnagotchi in auto mode
- sudo touch /root/.pwnagotchi-auto
- sudo systemctl restart pwnagotchi
And that all worked. But, now when my Pwnagotchi is in auto mode, it tries to connect to my iPhone, but it says 'NF' and I do not understand why. I've done this multiple times, with it connected, without, with my hotspot on, without, etc. but nothing ever works.

operationairstrike, 2024-01-16:
@evilsocket

studiozeroseven, 2024-08-01:
So I had this same issue with the iPhone, this was how I resolved it:

boot the RPi0W using the DATA PORT
Connected to the RPi0W with putty
Paired the iPhone with the RPi0W with these steps:

sudo bluetoothctl
scan on
put the iPhone in discovery mode (settings - bluetooth)
~~wait until the RPi0W found the iPhone then~~
In bluetoothctl enter: `discoverable on`
then my phone can see my pwnagotci and I tap on it on the iPhone and then tap PAIR
Go back to blurtoothctl and there will be a few (2 for me) times I had to type `yes` to pairing questions
stop the scanning process scan off
exit form bluetoothctl exit
now in the BT paired device list of the iPhone I can see the RPi0W (the hostname)
start the pwnagotchi in auto mode
sudo touch /root/.pwnagotchi-auto
Then reboot and boom it eventually connected after booting up

[image]