evilsocket/pwnagotchi #965: Support of WEP Networks

PwnPeter, 2021-03-04:
I think the support of the WEP protocol would be a significant plus.

qkum, 2021-03-11:
1# They are more or less non-existent in 2021 in the Western world. 

2# You are likely new to pentest/hacking game. One big problem with WEP is that if you attack a WEP network...you always end up with the password of the wifi. Not just an encrypted password. You get the REAL UN-ENCRYPTED PASSWORD = illegal in most countries.

PwnPeter, 2021-03-11:
1# I still see many in France

2# Pwnagotchi uses deauthentication attack to grab the handshake which is illegal without authorizationss in all contries.
So for me this is not a valid argument for not integrating WEP support.

PwnPeter, 2021-03-11:
> 1# They are more or less non-existent in 2021 in the Western world.
> 
> 2# You are likely new to pentest/hacking game. One big problem with WEP is that if you attack a WEP network...you always end up with the password of the wifi. Not just an encrypted password. You get the REAL UN-ENCRYPTED PASSWORD = illegal in most countries. Or in other words having WEP 5000 packets on your pc is just as good as the password. You can 100% break the encryption in seconds in 2021. So it is a nono legally wise.

1# I still see many in France

2# Pwnagotchi uses deauthentication attack to grab the handshake which is illegal without authorizationss in all contries.
So for me this is not a valid argument for not integrating WEP support.

qkum, 2021-03-21:
> > 1# They are more or less non-existent in 2021 in the Western world.
> > 2# You are likely new to pentest/hacking game. One big problem with WEP is that if you attack a WEP network...you always end up with the password of the wifi. Not just an encrypted password. You get the REAL UN-ENCRYPTED PASSWORD = illegal in most countries. Or in other words having WEP 5000 packets on your pc is just as good as the password. You can 100% break the encryption in seconds in 2021. So it is a nono legally wise.
> 
> 1# I still see many in France
> 
> 2# Pwnagotchi uses deauthentication attack (DOS) to grab the WPA handshake, which is illegal without authorizations in all contries.
> So for me this is not a valid argument for not integrating WEP support.
> Just add an option to disable it like the deauthentication option and I don't see the problem :)

You are crazy man. 

If you sniff the 5000 packets as the software automatically does. You litterly got their password. You broke in by catching 5000 packets from the air. 

WPA - everything not 10 year old WEP = You dont got their password even if you had 1 million of handshakes. It is harmless that you got encryped handshakes. You have not broken in. 

You are going to jail with that mentality. 

It is 100% on purpose that pwnagotchi does not attack WEP networks.

jesseuwu, 2021-03-26:
There is a perfectly legitimate use of including WEP support: convincing people who still run outdated WEP networks to switch off of WEP is so much easier if you can demonstrate to them just how easy it is to get their password. "Hey boss, check this out, all I gotta do is literally plug this thing in for 30 seconds and our Wifi password comes up. It's time to upgrade."

You can use a WEP sniffer in legitimate ways, just as you can use a WPA sniffer in an illegal way. The only difference is the strength of the cryptography. It's not a good argument not to include it as a feature.

I think pwnagotchi doesn't support it because it's uncommon to see WEP networks nowadays, but it should be included if not for any other reason, than to be feature-complete. You can always disable it if you're uncomfortable using it, and if you're not in an area where it's illegal for you to sniff the traffic, you shouldn't be using a pwnagotchi there anyways, whether or not it has the WEP feature enabled.

uwuforever, 2021-03-26:
There is a perfectly legitimate use of including WEP support: convincing people who still run outdated WEP networks to switch off of WEP is so much easier if you can demonstrate to them just how easy it is to get their password. "Hey boss, check this out, all I gotta do is literally plug this thing in for 30 seconds and our Wifi password comes up. It's time to upgrade."

You can use a WEP sniffer in legitimate ways, just as you can use a WPA sniffer in an illegal way. The only difference is the strength of the cryptography. It's not a good argument not to include it as a feature.

I think pwnagotchi doesn't support it because it's uncommon to see WEP networks nowadays, but it should be included if not for any other reason, than to be feature-complete. You can always disable it if you're uncomfortable using it, and if you're not in an area where it's illegal for you to sniff the traffic, you shouldn't be using a pwnagotchi there anyways, whether or not it has the WEP feature enabled.