evilsocket/pwnagotchi #974: Does Not switch to AI Mode

KatzeMau, 2021-03-26:
My pwnagotchi sadly does Not to AI Mode 
I got two pwnagotchis but sadly my second one does Not switch to AI Mode

Hope you can help me!

Happy Pwning from germany

noddynz, 2021-04-24:
Apparently installing 1.5.3, then letting it auto upgrade to 1.5.5 fixes the issue.

carlneuhaus, 2021-05-08:
The following line was in the log file when the AI tried to start up
` error while starting AI (numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject)`
Removing numpy and resinstalling seems to have fixed the issue on 1.5.5
```
pip3 uninstall numpy
pip3 install numpy
```

Crashcart, 2021-05-19:
I found a Reddit post suggesting that last night so I tried it. It complained about version numbers of numpy and pillow so I fixed that. But I get the same errors
`

[2021-05-19 19:30:16,384] | [ERROR] | error while starting AI (numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject)
-- | -- | --
&nbsp; | &nbsp; | ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject

`

Crashcart, 2021-05-19:
Yep did it for me. Thanks. Donno why I was resisting the idea in my head. Not like I couldn't roll back

KatzeMau, 2021-05-20:
Yes it worked for me too, could it be that it is coruptet during the installation?
Can i close the issue ?

aspman, 2021-05-20:
evilsocket is aware of the issue, as it appears to be something wrong with the build itself. Could probably do with being noted publicly until an updated image can be released (the requirement has been sorted in the repo from what I've seen, it's just not in the latest image).
So if the workaround has worked, I can't see why it would be a problem closing this thread.
(I'm quite new to git and build environments, so I'm not sure on etiquette 😅)

slabua, 2021-08-14:
I solved it by installing libatlas3-base.

JettScythe, 2021-09-11:
I was unable to fix with installing the latest numpy. Not sure if it matters, but I wasnt able to download on the pi (and I forgot to uninstall numpy, but I assumed it would overwrite) so I downloaded on my desktop, moved it to the pi and installed it with pip3 there. 

Will try with libatlas tomorrow

JettScythe, 2021-09-11:
Uninstalled both versions of numpy, reinstalled the latest and kept getting an error saying no Module "numpy" was found - got frustrated and am now just trying 1.5.3 instead - will hopefully auto update from there.

JettScythe, 2021-09-11:
I can confirm that installing 1.5.3 and upgrading to 1.5.5 acts as a workaround

0x0lli, 2021-09-12:
how did you guys configure the pi to have an outgoing connection from the PAN to the network on which your PC is on?
I am stuck at JettScythe's spot, that is, I cannot update or install anything. uninstalling numpy is fine, but I ran the windows-.ps1 script and also, tried to configure the network device settings. I get "IP-settings cannot be saved, check one or more settings and try again" (translated) once I try to enter the manual values from https://pwnagotchi.ai/configuration/#connect-to-your-pwnagotchi . I have a shell running, so everything worked until there.
I do not get an error running the ps1-script in adminmode-powershell, so I assume it did _some_ work, but it doesnt do the trick for me.

0x0lli, 2021-09-13:
https://www.youtube.com/watch?v=cnmrKCBzDRU
https://community.pwnagotchi.ai/t/mini-guide-bluetooth-tether-on-android/248
https://wiki.edgarbv.com/index.php/Pwnagotchi4

I have finally succeeded in setting up a bt-connection via bt-PAN but due to the countless trials needed, I have no idea how to reproduce the steps in order to write it down. the info on the homepage was not of any help to me, so in case anyone is in the same spot as I was, you might take a look into the link above.

I dont know when or if an auto update worked, as I additionally ran ```sudo pip3 install --upgrade pwnagotchi``` . There was no hint as to how to upgrade pwnagotchi manually and within the webcfg there is nothing but a generic info about auto-update but nothing else. Finally, 1.5.5 is running and AI working, startiung from a fresh 1.5.3.

KatzeMau, 2021-09-13:
Summary of solutions:
First you should try to update the device completely
```
sudo apt update && apt upgrade -y
sudo pip3 install --upgrade pwnagotchi
```
Restart and see if the problem has been resolved, otherwise reinstall "numpy" and install "libatlas3-base"
```
pip3 uninstall numpy
pip3 install numpy
 sudo apt install libatlas3-base -y
```

If the problem persists on your site after doing said things, open a new issue and mention issue 974 (This one)

Stay healthy :) 
And of course the most important thing happy pwning !

closed 2021-09-13 by KatzeMau

peterus, 2021-10-13:
this issue is STILL existing (one month later) with the current release. would it be possible to fix this in the code/repo and create a new release? for pawnagotchi newbies like me it is hard to fix issues like this.

databonanza, 2022-06-16:
Yep, still broken.

databonanza, 2022-06-16:
@evilsocket

lcrince, 2022-06-21:
I think it will stay broken for a while, there is also still no Waveshare v3 support and it is becoming increasingly impossible to buy the V2. 

To me it seems increasingly likely the project will become depreciated.

KatzeMau, 2022-06-21:
Sadly, it looks like it will become depreciated.
Adding support for the V3 display is not a real problem, but the project itself has a lot of small bugs that affect the overall experience.

peterus, 2022-06-21:
If there is interest, I can support in building up a build process which is building an image.

But I think this original repo is dead and should not be taken as a workplace.

slabua, 2022-06-22:
Been dead since a while, PR are simply ignored :\

evilsocket, 2022-06-22:
PRs are not ignored if i'm sure they don't further break the build and properly reference the issue they're fixing.

Ravi-099, 2023-02-28:
For anyone having an issue with the Waveshare V3 display, you can use the below config line to fix the problem.