evilsocket/pwnagotchi #975: Can't see this network in Windows 10

johnnyringo505, 2021-03-28:
I set up the pwanagotchi and it is getting handshakes with no problem.  However when I plug it in via USB into my computer using Windows 10 it is not recognizing it as a network and when I try to ssh it says the connection timed out.  How do I fix this?  I would like to retrieve the handshakes.

aspman, 2021-03-28:
I'm going to presume that this is an up-to-date version of W10, which broke rndis last year (I think that's the acronym).
In order to get it (the raspberry pi) to be seen correctly and function as a USB gadget, you'll have to obtain a rndis driver from a third party, as Microsoft removed it from the OS for some reason.
I don't have a link handy, but I believe the one I got came from a music creators forum for Korg device users. Googling "rndis windows Korg" should hopefully help you in your search.
I'll try and find the time to collaborate with the Devs of this project on updating the Windows host doc if this helps you.

uwuforever, 2021-03-30:
Here's the driver: https://modclouddownloadprod.blob.core.windows.net/shared/mod-rndis-driver-windows.zip

And how to install it:
1) Unzip the driver zip folder.
2) Plug in Pwnagotchi
3) Go to your Device Manager
4) Find your Pwnagotchi under "Ports (COM & LPT)", it'll be one of the "USB Serial Device (COM#)". It might help here to disconnect your other USB devices.
5) Right click that device, go to "Update drivers", choose "Browse my computer for drivers", then browse to the folder you unzipped in #1.
6) Click next through remaining dialogs to install the driver.

Should work now.

DarthKegRaider, 2021-08-23:
Thanks uwuforever, drivers worked a treat!  I can now connect to my faceless pwnagotchi!  Better buy him a face soon, i'm loving this project.

sokeman, 2022-11-22:
I'm having the same problem with win11 with the pwnagotchi staying in auto mode and nothing is being picked up by the computer. There's no sound when plugged in and it doesn't show in device manager. any ideas???

Hieuhuule, 2022-12-30:
Same issue on my Windows Surface Laptop 4 running Windows 10 home and using a Raspberry Pi Zero W v1.1

Flashes successfully using the following image that patches the Waveshare V3, https://archive.org/details/pwnagotchi_1.5.5_WSV3Patched

However, using the microUSB cable that came with my Chromecast which allows for data transfer does not cause it to boot in MANUAL mode. It is in the correct port and I have tested other cables just in case, all to no avail.

Upon checking my DEVICE MANAGER, I do not see any option for RNDIS or any unidentified USB Device that would point to my pwnagotchi. So there is no obvious choice to UPDATE DRIVER with the files from uwuforever. I'm at a loss and not sure how to proceed with current on hand equipment.

timmosaurusrex, 2022-12-30:
What i did to make it work was to check for optional updates and select the ethernet gadget or something named like that.

NachiThePanda, 2023-03-09:
> I'm having the same problem with win11 with the pwnagotchi staying in auto mode and nothing is being picked up by the computer. There's no sound when plugged in and it doesn't show in device manager. any ideas???

I am having the same issue. I went ahead and made a post #1145 about it. Hopefully someone has a fix or solution. For now I would just try a VM or borrowing a computer to SSH

IcedOdin, 2024-03-13:
https://joe.blog.freemansoft.com/2022/11/installing-rndis-driver-on-windows-11.html this tells you how to activate the once built in driver which works just fine