evilsocket/pwnagotchi #976: Ability to enable/disable Pwnagotchi announcements

uwuforever, 2021-03-30:
**Is your feature request related to a problem? Please describe.**
There are tools such as: https://www.nzyme.org/post/nzyme-v1-0-0-kyle-canyon-has-been-released that detect Pwnagotchi through its advertisement protocol. It's a cute feature, but it would be nice to be able to turn it off when it's not needed to avoid tracking.

**Describe the solution you'd like**
An option, maybe enabled through the plugins toggle page, that would allow a user to disable Pwnagotchi friend announcements so it is not so easily trackable.

thecoder-001, 2021-03-30:
Hello!
You can disable wifi advertising by editing your pwnagotchi's `config.toml` to include 
```toml
personality.advertise = true
```
as documented [here](https://pwnagotchi.ai/usage/#training-the-ai):
```toml
# advertise our presence
personality.advertise = true
```
Cheers

thecoder-001, 2021-03-31:
Hello again! If your issue was resolved,  it would be nice if you could close it to help evilsocket keeping the issue tracker tidy.
Cheers!

closed 2021-04-01 by uwuforever