evilsocket/pwnagotchi #977: [BUG]

maichai, 2021-03-30:
**Describe the bug**

aircrackonly does not get executed for every pcap that is collected in /root/handshakes

**To Reproduce**
Steps to reproduce the behavior:
1. install aircrack-ng and aircrackonly plugin and enable it
2. reboot
3. wait
4. check pcaps in /root/hashshakes using wireshark and identify pcaps that contain no packets or that contain only handshake message 1, not 2, 3, or handshake message 4 datagrams
5. check the log file whether there is output from aircrackonly for those pcaps 
6. see that there is no output from aircrackonly for those pcaps
7. deduce that aircrackonly was not executed for those pcaps, because it always generates an output for every pcap which it processes

**Expected behavior**
aircrackonly plugin should be called for every pcap collected in /root/handshakes, there should be an aircrackonly plugin output for each pcap in the log file, and "uncrackable" pcaps should be removed from the folder

**Screenshots**
n/a

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.3
 - OS version Linux pwnagotchi 4.19.93-Re4son+ #1 Wed Jan 8 14:55:22 AEDT 2020 armv6l GNU/Linux
 - Type of hardware Pi0 W with waveshare v2
 - Any additional hardware used: -

uwuforever, 2021-03-30:
I think this belongs as an issue on the custom plugins repo, not this repo, since this is a bug with one of the custom plugins and not a bug in pwnagotchi itself.

maichai, 2021-03-30:
I think you musunderstand my post. I am saying that the problem is in the plugin mechanism, and not in the plugin.