evilsocket/pwnagotchi #979: Can anyone help me with my setup with the inkyphat v2 ????

lildaze, 2021-04-04:
can't get it to display anything

xhrzg2017, 2021-04-06:
> 无法让它显示任何东西

我短接RST和BUSY就出现画面

noddynz, 2021-04-20:
Having the same issue on a freshly re-flashed image. What it has done though that the previous version didn't is it cleared the display to white from the InkyPhat logo. Got my hopes up haha.

ahaltric, 2021-04-22:
this works for me on both latest and 1.5.3 images although on the latest image AI is not working and I did not have time to dig it. 

so...
get pwnagotchi-raspbian-lite-v1.5.3.img (the AI is crashing on the latest pwnagotchi-raspbian-lite-v1.5.5.img)

connect to your  PC or with an external NIC to the Internet and allow pwnagotchi to auto-update ... 
will install pwnagotchi 1.5.5

after pwnagotchi got updated to 1.5.5... 
install the latest inky library 1.2.0 ( pwnagotchi will install 0.0.5 ) 

sudo pip3 install inky=1.2.0 
(will throw a warning that pwnagotchi requires inky 0.0.5 - ignore that)

disable pwnagotchi auto update for now until the dev team will update to newest  inky library.
I've tested ... installed inky=1.2.0 on pwagotchi 1.5.3 and after self update to 1.5.5 it removed inky 1.2.0 and installed back 0.0.5...  

modify inky.py from (/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/inky.py)

    def initialize(self):
        logging.info("initializing inky display")

        if self.config['color'] == 'black':
            from inky.auto import auto
            self._display = auto(ask_user=True, verbose=True)
            self._display.set_border(self._display.BLACK)
        else:
            from inky.auto import auto
            self._display = auto()
            self._display.set_border(self._display.BLACK)
            self._layout['width'] = self._display.WIDTH
            self._layout['height'] = self._display.HEIGHT
    

Hope it helps.

Gadgetoid, 2021-04-23:
Looks like pwnagotchi needs a version bump to grab the latest Inky library, then? I had not anticipated this since I couldn't go through the install process.

Is it sufficient just to update requirements.txt? - https://github.com/evilsocket/pwnagotchi/blob/master/requirements.txt

closed 2021-04-23 by evilsocket

noddynz, 2021-04-24:
> this works for me on both latest and 1.5.3 images although on the latest image AI is not working and I did not have time to dig it.
> 
> so...
> get pwnagotchi-raspbian-lite-v1.5.3.img (the AI is crashing on the latest pwnagotchi-raspbian-lite-v1.5.5.img)
> 
> connect to your PC or with an external NIC to the Internet and allow pwnagotchi to auto-update ...
> will install pwnagotchi 1.5.5
> 
> after pwnagotchi got updated to 1.5.5...
> install the latest inky library 1.2.0 ( pwnagotchi will install 0.0.5 )
> 
> sudo pip3 install inky=1.2.0
> (will throw a warning that pwnagotchi requires inky 0.0.5 - ignore that)
> 
> disable pwnagotchi auto update for now until the dev team will update to newest inky library.
> I've tested ... installed inky=1.2.0 on pwagotchi 1.5.3 and after self update to 1.5.5 it removed inky 1.2.0 and installed back 0.0.5...
> 
> modify inky.py from (/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/inky.py)
> 
> ```
> def layout(self):
>     fonts.setup(10, 8, 10, 28, 25, 9)
>     self._layout['width'] = 250
>     self._layout['height'] = 122
>     self._layout['face'] = (0, 40)
>     self._layout['name'] = (5, 18)
>     self._layout['channel'] = (0, 0)
>     self._layout['aps'] = (30, 0)
>     self._layout['uptime'] = (185, 0)
>     self._layout['line1'] = [0, 12, 250, 12]
>     self._layout['line2'] = [0, 110, 250, 110]
>     self._layout['friend_face'] = (0, 94)
>     self._layout['friend_name'] = (40, 96)
>     self._layout['shakes'] = (0, 111)
>     self._layout['mode'] = (225, 111)
>     self._layout['status'] = {
>         'pos': (120, 18),
>         'font': fonts.status_font(fonts.Small),
>         'max': 20
>     }
>     return self._layout
> ```
> 
> and
> 
> ```
> def initialize(self):
>     logging.info("initializing inky display")
> 
>     if self.config['color'] == 'black':
>         from inky.auto import auto
>         self._display = auto()
>         self._display.set_border(self._display.BLACK)
>     else:
>         from inky.auto import auto
>         self._display = auto()
>         self._display.set_border(self._display.BLACK)
>         self._layout['width'] = self._display.WIDTH
>         self._layout['height'] = self._display.HEIGHT
> ```
> 
> Hope it helps.

Could you please post your whole inky.py file? I for the life of me still cant get this to work.

bdgoodman, 2021-12-23:
As of this comment, setting `ui.display.color` to `auto` correctly init the display.

As an update to the suggestion from @noddynz here is my layout:

```
def layout(self):
        fonts.setup(10, 8, 10, 28, 25, 9)
        self._layout['width'] = 250 #212
        self._layout['height'] = 122 #104
        self._layout['face'] = (0, 40) #(0, 37)
        self._layout['name'] = (5, 18)
        self._layout['channel'] = (0, 0)
        self._layout['aps'] = (30, 0)
        self._layout['uptime'] = (185, 0) #(147, 0)
        self._layout['line1'] = [0, 12, 250, 12] #[0, 12, 212, 12]
        self._layout['line2'] = [0, 110, 250, 110] #[0, 92, 212, 92]
        self._layout['friend_face'] = (0, 104) #(0, 76)
        self._layout['friend_name'] = (40, 106) #(40, 78)
        self._layout['shakes'] = (0, 111) #(0, 93)
        self._layout['mode'] = (255, 111) #(187, 93)
        self._layout['status'] = {
            'pos': (135, 37), #(102, 18),
            'font': fonts.status_font(fonts.Small),
            'max': 20
        }
        return self._layout
```

Separately, refreshes happen at a rate that are too frequent for this display, so I modified to throttle that to a max of one refresh every 10 seconds.

```    def __init__(self, config):
        super(Inky, self).__init__(config, 'inky')
        self._display = None
        self._lastRefresh = 0
```

and

```
    def render(self, canvas):
        if self.config['color'] == 'black' or self.config['color'] == 'fastAndFurious':
            display_colors = 2
        else:
            display_colors = 3

        img_buffer = canvas.convert('RGB').convert('P', palette=1, colors=display_colors)
        if self.config['color'] == 'red':
            img_buffer.putpalette([
                255, 255, 255,  # index 0 is white
                0, 0, 0,  # index 1 is black
                255, 0, 0  # index 2 is red
            ])
        elif self.config['color'] == 'yellow':
            img_buffer.putpalette([
                255, 255, 255,  # index 0 is white
                0, 0, 0,  # index 1 is black
                255, 255, 0  # index 2 is yellow
            ])
        else:
            img_buffer.putpalette([
                255, 255, 255,  # index 0 is white
                0, 0, 0  # index 1 is black
            ])

        self._display.set_image(img_buffer)
        try:
            currentTime = time.time()

            if((self._lastRefresh == 0) or ((self._lastRefresh+10) <= currentTime)):
                self._display.show()
                self._lastRefresh = time.time()
        except:
            logging.exception("error while rendering on inky")

```

With config.toml having `ui.fps = 0.1
[inky.zip](https://github.com/evilsocket/pwnagotchi/files/7771617/inky.zip)
`

Mitkah, 2022-03-24:
> self._layout['mode'] = (255, 111) #(187, 93)

I think there is a bug, `self._layout['mode'] = (225, 111) `should be right to display the mode

JWhiteUX, 2022-03-26:
Any idea when we can expect InkyPhat support? I tried all of the above and nothing seems to work.

commit a9f07e9f (2021-04-23): fix: version bumped inky to 1.2.0 (fixes #979)