evilsocket/pwnagotchi #980: [BUG] Inky pHAT

handleprobereq, 2021-04-07:
Hello,
I've just gotten my new pi and display, tried to run it and it doesn't turn on. I did try to change the config, still nothing.

Display:
Black/White Inky pHAT (ePaper/eInk/EPD) [PIM-368]

**Any help will be appreciated!**

```
main.name = "Pwn"
main.lang = "en" main.whitelist =
[
"PRIVACY REASONS",
"PRIVACY REASONS",
"PRIVACY REASONS"
]
main.plugins.grid.enabled = true                                                                                                                                           main.plugins.grid.report = true
main.plugins.grid.exclude = [
"PRIVACY REASONS",
"PRIVACY REASONS",
"PRIVACY REASONS"
] 
ui.display.enabled = true
ui.display.type = "inky"
ui.display.color = "black"
```

Gadgetoid, 2021-04-09:
I think this is related to: https://github.com/evilsocket/pwnagotchi/issues/941

You can use `python3 -m inky.eeprom` to confirm your display variant. I don't know if the SSD1608 versions - ID's 10, 11 and 12 - are supported yet.

Gadgetoid, 2021-04-09:
Adding support for auto detection could help this, and might look something like:

```patch
diff --git a/pwnagotchi/ui/hw/inky.py b/pwnagotchi/ui/hw/inky.py
index bb02fd7..7f71b7a 100644
--- a/pwnagotchi/ui/hw/inky.py
+++ b/pwnagotchi/ui/hw/inky.py
@@ -42,6 +42,12 @@ class Inky(DisplayImpl):
             from pwnagotchi.ui.hw.libs.inkyphat.inkyphatfast import InkyPHATFast
             self._display = InkyPHATFast('black')
             self._display.set_border(InkyPHATFast.BLACK)
+        elif self.config['color'] == 'auto':
+            from inky.auto import auto
+            self._display = auto()
+            self._display.set_border(self._display.BLACK)
+            self._layout['width'] = self._display.WIDTH
+            self._layout['height'] = self._display.HEIGHT
         else:
             from inky import InkyPHAT
             self._display = InkyPHAT(self.config['color'])
```

There's already precedent for not-exactly-a-colour choices here; the 2-colour `fastAndFurious` mode.

In all cases it might be safer to run auto detection first so an error can be thrown for the SSD1608 pHAT boards which will be incompatible with the `fastAndFurious` mode anyway.

PR #985 (2021-04-18): Add support for SSD1608 variant 250x122 pixel Inky pHATs
## Description

This patch adds support for auto-detecting and using them when "ui.display.color" is set to "auto".

## Motivation and Context

We (that is, Pimoroni) released a set of Inky display variants which are virtually indistinguishable from their predecessors visually, but functionally incompatible due to a change of driver IC and resolution. I gave a heads up in #941.

We include an EEPROM to ID our boards. Running `python3 -m inky.eeprom` should show display variant. Variants 10, 11 and 12 are SSD1608 based. This change uses t [... cut]
commit 34c2c8a0 (2021-04-18): Add support for SSD1608 variant 250x122 pixel Inky pHATs

Running `python3 -m inky.eeprom` should show display variant. Variants 10, 11 and 12 are SSD1608 based.

This patch adds support for auto-detecting and using them when "ui.display.color" is set to "auto".

Fixes #980 and #941
closed 2021-04-18 by evilsocket